WebODM/WebODM #232: Webodm ortophoto problem

bejixster, 2017-07-09:
I put link that explain better my problem.
https://www.youtube.com/watch?v=36WGAY9TtOU

kikislater, 2017-07-09:
I think it's better describing than whatching a long video ...

bejixster, 2017-07-09:
I am sorry for the long video but i think it's the best way to explain .
I am only a technician  i work as a surveyor so my programming skills are Zero :-)

First 
-drone phantom3 4k took the pictures using dronedeploy
-georeference with gps rtk points locally 
-create file gplist with help of qgis (wgs84 utm 29N)
-insert the pictures and gcpoint list on webodm
-I Use default options (maybe i have to change something but i don't know what)
-It was complete with this error
**An orthophoto could not be generated. To generate one, make sure GPS information is embedded in the EXIF tags of your images.**
-In the view 3d model the cloud point its up side down

My gcp_list
[gcp_list.txt.txt](https://github.com/OpenDroneMap/WebODM/files/1133981/gcp_list.txt.txt)

[webodm_process.txt](https://github.com/OpenDroneMap/WebODM/files/1133986/webodm_process.txt)

pierotofy, 2017-07-09:
From the log this stands out:

```
Error in Georef:
Fewer than 3 GCPs have correspondences in the generated model.
```

So in short, something didn't work out in the georeferencing step. As a result, the point cloud looks upside down because there's no georeferencing information, so the model is displayed in local coordinates.

I'm not too familiar with the Georef module, perhaps @dakotabenjamin has seen this problem before?

@bejixster could you share your dataset for others to try to process?

bejixster, 2017-07-09:
https://drive.google.com/drive/folders/0B83WgjZuQM1MeFBmMmZjeUJIeDg?usp=sharing

pierotofy, 2017-07-12:
This might be related: https://github.com/OpenDroneMap/OpenDroneMap/issues/541

pierotofy, 2017-11-25:
Ok the error is due to the format of the gcp file. Use “JPG” instead of “jpg” in the file extensions.

closed 2017-11-25 by pierotofy