WebODM/WebODM #233: Fail when uploading more than 2k images

patrickelectric, 2017-07-10:
[image]
[image]
[image]

```
IMG_170612_162308_0000_GRE.TIF
IMG_170612_162308_0000_NIR.TIF
IMG_170612_162308_0000_RED.TIF
IMG_170612_162308_0000_REG.TIF
IMG_170612_162308_0000_RGB.JPG
IMG_170612_162310_0001_GRE.TIF
IMG_170612_162310_0001_NIR.TIF
IMG_170612_162310_0001_RED.TIF
IMG_170612_162310_0001_REG.TIF
IMG_170612_162310_0001_RGB.JPG
IMG_170612_162312_0002_GRE.TIF
IMG_170612_162312_0002_NIR.TIF
IMG_170612_162312_0002_RED.TIF
IMG_170612_162312_0002_REG.TIF
IMG_170612_162312_0002_RGB.JPG
IMG_170612_162313_0003_GRE.TIF
IMG_170612_162313_0003_NIR.TIF
IMG_170612_162313_0003_RED.TIF
IMG_170612_162313_0003_REG.TIF
IMG_170612_162313_0003_RGB.JPG
IMG_170612_162315_0004_GRE.TIF
IMG_170612_162315_0004_NIR.TIF
IMG_170612_162315_0004_RED.TIF
IMG_170612_162315_0004_REG.TIF
IMG_170612_162315_0004_RGB.JPG
IMG_170612_162316_0005_GRE.TIF
IMG_170612_162316_0005_NIR.TIF
IMG_170612_162316_0005_RED.TIF
IMG_170612_162316_0005_REG.TIF
IMG_170612_162316_0005_RGB.JPG
IMG_170612_162318_0006_GRE.TIF
IMG_170612_162318_0006_NIR.TIF
IMG_170612_162318_0006_RED.TIF
IMG_170612_162318_0006_REG.TIF
IMG_170612_162318_0006_RGB.JPG
IMG_170612_162320_0007_GRE.TIF
IMG_170612_162320_0007_NIR.TIF
IMG_170612_162320_0007_RED.TIF
IMG_170612_162320_0007_REG.TIF
IMG_170612_162320_0007_RGB.JPG
IMG_170612_162322_0008_GRE.TIF
IMG_170612_162322_0008_NIR.TIF
IMG_170612_162322_0008_RED.TIF
IMG_170612_162322_0008_REG.TIF
IMG_170612_162322_0008_RGB.JPG
IMG_170612_162324_0009_GRE.TIF
IMG_170612_162324_0009_NIR.TIF
IMG_170612_162324_0009_RED.TIF
IMG_170612_162324_0009_REG.TIF
IMG_170612_162324_0009_RGB.JPG
IMG_170612_162326_0010_GRE.TIF
IMG_170612_162326_0010_NIR.TIF
IMG_170612_162326_0010_RED.TIF
IMG_170612_162326_0010_REG.TIF
IMG_170612_162326_0010_RGB.JPG
IMG_170612_162328_0011_GRE.TIF
IMG_170612_162328_0011_NIR.TIF
IMG_170612_162328_0011_RED.TIF
IMG_170612_162328_0011_REG.TIF
IMG_170612_162328_0011_RGB.JPG
IMG_170612_162330_0012_GRE.TIF
IMG_170612_162330_0012_NIR.TIF
IMG_170612_162330_0012_RED.TIF
IMG_170612_162330_0012_REG.TIF
IMG_170612_162330_0012_RGB.JPG
[... 2000 lines cut ...]
IMG_170612_163654_0413_GRE.TIF
IMG_170612_163654_0413_NIR.TIF
IMG_170612_163654_0413_RED.TIF
IMG_170612_163654_0413_REG.TIF
IMG_170612_163654_0413_RGB.JPG
IMG_170612_163656_0414_GRE.TIF
IMG_170612_163656_0414_NIR.TIF
IMG_170612_163656_0414_RED.TIF
IMG_170612_163656_0414_REG.TIF
IMG_170612_163656_0414_RGB.JPG
IMG_170612_163659_0415_GRE.TIF
IMG_170612_163659_0415_NIR.TIF
IMG_170612_163659_0415_RED.TIF
IMG_170612_163659_0415_REG.TIF
IMG_170612_163659_0415_RGB.JPG
```
Computer:
```
OS: Arch Linux 
Kernel: x86_64 Linux 4.11.6-3-ARCH
CPU: Intel Core i3-7100 @ 4x 3.9GHz [27.8°C]
GPU: GeForce GTX 1050 Ti
RAM: 4116MiB / 7940MiB
SWAP: 5239MiB / 20000MiB
```

kikislater, 2017-07-10:
Here is a Parrot sequoia, you try to upload 3 band RGB 16mpix and 4 x 1 band 1.2mpix at the same time. 
Also did you apply vignetting correction and reflectance estimation of each band before processing ?

patrickelectric, 2017-07-10:
@kikislater I am doing this in the same time to see if it's possible to process all bands with the RGB exif info.
@kikislater I will add this configuration now, but do you know how to realize the orthomosaic with the others bands that don't have exif gps data ?

kikislater, 2017-07-10:
It's a bad idea to process RGB and 4 bands. What you should expect, no way to do that ! I don't understand and if you are a gis user it's a non sense to doing it.

If you don't have GPS in exif with 4 bands, there is a problem during acquisition. Low power of sequoia. 
I use other software at this time to process it, but I'm starting a project few weeks ago to process it with opensource software. Could be fastest ... time will say it !

pierotofy, 2017-07-10:
I'm pretty sure the error is caused by the low default `DATA_UPLOAD_MAX_NUMBER_FIELDS` value (which is set to 1000).

I will increase this to 10000 for the next release, in the meanwhile, you might want to add the following to your `webodm\settings.py` file and rebuild webodm (if using docker) or restart it if running natively:

```
DATA_UPLOAD_MAX_NUMBER_FIELDS=10000
```

patrickelectric, 2017-07-10:
@kikislater 
> It's a bad idea to process RGB and 4 bands. What you should expect, no way to do that ! I don't understand and if you are a gis user it's a non sense to doing it.
Pix4D does that.. I was just testing it with ODM.
This project that ou talked about is in GitHub ? Can you share it here ?

Wow, @pierotofy tys !

pierotofy, 2017-07-10:
Note that even after increasing `DATA_UPLOAD_MAX_NUMBER_FIELDS` I'm still having other issues with larger than usual datasets. I will post updates when I know more.

kikislater, 2017-07-10:
What you have after with pix4d ? An RGB ortho and a multispectral ortho ? If it's that, there is a simply script to separate datas and process but afaik ODM does not support it ! 
Not at this time, project is in my computer ^_^ but I will push it on ODM or QGIS ...

pierotofy, 2017-07-10:
Besides setting a higher value for `DATA_UPLOAD_MAX_NUMBER_FIELDS`, the problem goes all the way down to the operating system...

```bash
# ulimit -a
core file size          (blocks, -c) unlimited
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 95659
max locked memory       (kbytes, -l) 64
max memory size         (kbytes, -m) unlimited
open files                      (-n) 1024
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 0
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 95659
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited
```

The open files limit sets a cap on the number of images that can be currently uploaded with the current code. It's possible to increase this limit by running:

```bash
# ulimit -n 3000
```

I'm currently evaluating possible workarounds to this issue.

patrickelectric, 2017-07-10:
@pierotofy It's possible to upload 1k photos, and than when clicking in upload again, it will upload more 1k photos if the user select to append the actual project photos with the one that he want to upload.

pierotofy, 2017-07-10:
We can do better than that :)

These new changes https://github.com/OpenDroneMap/WebODM/pull/230/commits/d376a1e779bc197aa01a95b435b91d9ebec72104 allow WebODM to handle a lot more files, now capped by the amount of RAM available on the computer and not by the "open files" OS limit.

In the future we will have to switch to a streamed approach to handle even larger file uploads.

Changes will be available in the next few days, as soon as #230 is merged in.

closed 2017-07-10 by pierotofy