WebODM/WebODM #234: Process exited code 1

marnold8, 2017-07-11:
I for the life of me can't get past this error code 1:

Traceback (most recent call last):
  File "/code/run.py", line 8, in <module>
    import ecto
  File "/code/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

pierotofy, 2017-07-11:
Hi @marnold8, we recommend using docker to install WebODM. How did you install ODM/WebODM?

marnold8, 2017-07-11:
I used the instructions on the github opendronemap/WebODM page
Should I start from the beginning?

pierotofy, 2017-07-11:
Yes please. Keep note of the steps you are taking, then report them here if the error persists. Also include the operating system you are on.

marnold8, 2017-07-11:
Thanks a bunch.

closed 2017-07-22 by marnold8