WebODM/WebODM #235: Question: Is it possible to change default port?

macsee, 2017-07-12:
Hi,
I have WebODM installed on a Ubuntu machine and I would like to use it from internet. I would usually do it from a PC that's in firewalled network, so I was wondering since 8000 port is blocked on this network, if it's possible to change it for another that's not blocked, like 80.
Thanks!

pierotofy, 2017-07-12:
Hi @macsee, yes, you just need to change this `docker-compose.yml` line:

From:

```
ports:
  - "8000:8000"
```

To:

```
ports:
  - "80:8000"
```

Then restart WebODM.

macsee, 2017-07-12:
Great! Thanks @pierotofy!

Cheers.

closed 2017-07-12 by pierotofy

deviirmr, 2019-01-29:
Hi,
I have a similar issue, I have change the default port to some port (8001:8000) as per @pierotofy but still the application is not opening in new port but  active in the old port.
[image]
I restart the application but seems no impect
[image]
Any sugustion.
Thanks.

pierotofy, 2019-01-29:
I would suggest not to modify the docker-compose.yml file and instead use the `--port` param when launching WebODM:

`./webodm.sh restart --port 8001`

deviirmr, 2019-01-29:
Finally, it is working.
Thank you so much and thanks for the quick response.
[image]