WebODM/WebODM #238: Orthophoto upside down.

marioriverachavarria, 2017-07-12:
The orthophoto is turned upside down.
This is the result:
[image]
And this is the one should be:
[image]
Where could be the problem?.

This is with the latest version of WebODM

dakotabenjamin, 2017-07-13:
It might occur if the photos don't have altitude data but I don't know without seeing the data

marioriverachavarria, 2017-07-13:
Hello,

Photos do have the height info , the problem is that this is interpreted wrongly
Heres an example dataset.

https://www.dropbox.com/sh/wd3dkzdhjrs0xia/AAAYHgZu_F_AOex8-3hRorHfa?dl=0

Thank you

marioriverachavarria, 2017-07-14:
:)

KommandorKeen, 2017-07-14:
What did you change to go to the :)  ?
I ask because I have had exactly the same problem in an area with big trees and the drone flying at a single height. All photos have altitude data.

marioriverachavarria, 2017-07-15:
its really a :( haha. No results yet. I maen everything appears to be fine but its just the inverse.

Weird.

KommandorKeen, 2017-07-15:
I solve the problem with mine. I had one extra image which I had taken at an angle, and a different altitude and bang it all flipped and fitted.
So my work around is to, from now on, capture an extra image or two on the drone descent with the camera at -70 deg from horizontal and th put them in the data set.
Simon

pierotofy, 2017-07-15:
Could you try to reprocess the dataset by increasing the `min-num-features` option to 8000 (up from the 4000 default)?

marioriverachavarria, 2017-07-15:
Same results.
[image]

Problem is that is getting the height data as negative. Is there a way to change the sign of the operation?

[image]

pierotofy, 2017-07-15:
Something does not look right with your EXIF tags. For example, YUN00095.jpg gives me:

```
#exiftool YUN00095.jpg
GPS Altitude                    : 5 m Below Sea Level
```

But I have a feeling that that's not correct. Check the validity of your altitude values, as they are probably incorrect, fix the exif tags or use a GCP file. The model will come out OK afterwards.

marioriverachavarria, 2017-07-15:
This happens also without using exif.

marioriverachavarria, 2017-07-15:
Is it possible to make the software ignore this attribute GPSAltitudeRef?

marioriverachavarria, 2017-07-15:
else:
            altref = '1'
        key = 'Exif.GPSInfo.GPSAltitudeRef'
        metadata[key] = pyexiv2.ExifTag(key, altref)

        # write values
        metadata.write()

closed 2017-09-01 by pierotofy