WebODM/WebODM #239: Files Missing and Other Errors on Native Installation

macsee, 2017-07-13:
Hello,
I've tryed to install WebODM on a P.C with a fresh installation of Ubuntu 16.04, following the steps for native installation, and I've got the following output when executing start.sh:

Checking python version... 3.x, good!
Checking GDAL version... GDAL 2.1.3, released 2017/20/01, excellent!
Running migrations
INFO Booting...
WARNING Debug mode is ON (for development this is OK)
INFO Starting background scheduler...
Operations to perform:
Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
Running migrations:
No migrations to apply.
WARN: /webodm/build/static does not exist, CSS, JS and other files might not be available.
nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (13: Permission denied)
2017/07/12 22:34:35 [warn] 17883#17883: the "user" directive makes sense only if the master process runs with super-user privileges, ignored in /home/macsee/WebODM/nginx/nginx.conf:4
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] still could not bind()
[2017-07-12 22:34:38 -0300] [17884] [INFO] Starting gunicorn 19.7.1
[2017-07-12 22:34:38 -0300] [17884] [INFO] Listening at: unix:/tmp/gunicorn.sock (17884)
[2017-07-12 22:34:38 -0300] [17884] [INFO] Using worker: sync
[2017-07-12 22:34:38 -0300] [17890] [INFO] Booting worker with pid: 17890

Congratulations! └@(･◡･)@┐

If there are no errors, WebODM should be up and running!

Open a web browser and navigate to http://localhost:8000

When I open a web browser, everything looks as if CSS and Javascript files were missing. What's the problem now? I have also tried executing this script: contrib/ubuntu_1604_install.sh but nothing different happens. I dont know what else to try. Can you help me please?
Thanks!

pierotofy, 2017-07-13:
Hi @macsee, you must have skipped this step:

```
pip install -r requirements.txt
sudo npm install -g webpack
npm install
webpack
python manage.py collectstatic --noinput
```

macsee, 2017-07-13:
Hi @pierotofy.
I'm pretty sure I did that. Furthermore, I had some errors when I executed `sudo npm install -g webpack`. I'll do it again, just to be sure.
A few questions. Those commands I ran them in a python virtual environment using python 3, is that allright? And another one. If I ran `the ubuntu_1604_install.sh` script, which are the following steps? Thanks!

pierotofy, 2017-07-13:
Each command should be executed only if the previous one executed without errors. If there's an error anywhere in the process, chances are things will not work! Yes using a Python 3 environment is correct. I would open the `ubuntu_1604_install.sh` script to see which steps are executed, I don't maintain that script so I'm not sure if it follows all the steps in the README.

macsee, 2017-07-13:
Ok. I'll wait for your comments about that script. Thanks!

macsee, 2017-07-14:
Hi @pierotofy,
I tried all over again. I Cloned the repo and execute the commands from the README. When I ran `sudo npm install -g webpack` I had a couple of warnings but not errors:

> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/webpack/node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

The same for `npm install`:

>npm WARN deprecated tween.js@16.6.0: This package has moved to @tweenjs/tween.js. Please update your dependencies.
npm WARN prefer global node-gyp@3.6.2 should be installed with -g
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

The other commands ran fine, but in the end I've always got the same problem.

I discovered that if I run `./start.sh --no-gunicorn` everything works fine, but if I run without the `--no-gunicorn` option, the output is this:

> WARN: /webodm/build/static does not exist, CSS, JS and other files might not be available.
nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (13: Permission denied)
2017/07/13 20:38:56 [warn] 10114#10114: the "user" directive makes sense only if the master process runs with super-user privileges, ignored in /home/macsee/WebODM/nginx/nginx.conf:4
[2017-07-13 20:38:56 -0300] [10117] [INFO] Starting gunicorn 19.7.1
[2017-07-13 20:38:56 -0300] [10117] [INFO] Listening at: unix:/tmp/gunicorn.sock (10117)
[2017-07-13 20:38:56 -0300] [10117] [INFO] Using worker: sync
[2017-07-13 20:38:56 -0300] [10123] [INFO] Booting worker with pid: 10123
Congratulations! └@(･◡･)@┐
==========================
If there are no errors, WebODM should be up and running!
Open a web browser and navigate to http://localhost:8000
NOTE: Windows users using docker should replace localhost with the IP of their docker machine's IP. To find what that is, run: docker-machine ip
INFO Booting...
WARNING Debug mode is ON (for development this is OK)
INFO Starting background scheduler...
WARNING Session data corrupted
WARNING Session data corrupted

It appears to be looking for files in the folder `/webodm/build/static`, but I don't have a `build` folder inside of `webodm`. That folder is in another location. Why? I tryed to manually copy that folder in the right path but for some reason is not working, I've got the same result.

I don't know what else to do.

pierotofy, 2017-07-14:
Just keep using `./start.sh --no-gunicorn`, it will not make much of a difference as long as you keep WebODM for your own use (ex. not available on the internet).

To use nginx you need to configure `nginx/nginx.conf` to match your system paths, the default one is meant to work on our docker images.

merkato, 2019-04-19:
Trying with native install on DigitalOcean droplet. Run into issue with webpack installation.

`npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})'

Googled for an answer, but I'm too dumb for npm modules ;)

pierotofy, 2019-04-19:
That's just a warning. Shouldn't affect the installation.