WebODM/WebODM #240: Process exited with code 1 - Merging depthmaps

husnimubarok, 2017-07-13:
I'm using docker setup. It failed once all images uploaded and start processing.

> Merging depthmaps
> Killed
> Traceback (most recent call last):
>   File "/code/run.py", line 46, in <module>
>     plasm.execute(niter=1)
>   File "/code/scripts/opensfm.py", line 128, in process
>     (context.pyopencv_path, context.opensfm_path, tree.opensfm))
>   File "/code/opendm/system.py", line 34, in run
>     raise Exception("Child returned {}".format(retcode))
> Exception: Child returned 137
>

pierotofy, 2017-07-13:
Hi @husnimubarok, WebODM should have displayed this error message (did it?)

```
It looks like your processing node ran out of memory. If you are using docker, make sure that your docker environment has enough RAM allocated.
```

http://stackoverflow.com/a/39720010
https://docs.docker.com/docker-for-windows/#advanced

husnimubarok, 2017-07-14:
Yes it appeared right below the error. It seems my container out of memory when processing 124 images. right now I'm using 3.9 GB limit. It was success generating ortho and dem for 63 images, is there any rule of thumbs on the resource requirement?

pierotofy, 2017-07-14:
Currently we simply go for "the more the better". In the next few months some improvements in the opensfm module will reduce memory usage by quite a bit, as well as speed up computation, so any numbers that we might benchmark today might not apply tomorrow. Once we reach the first stable release we will probably draft a document with guidelines.

closed 2017-07-14 by pierotofy