WebODM/WebODM #241: NDVI

pierotofy, 2017-07-13:
WebODM should support the creation/display of NDVI maps for use in precision agriculture. I'll open this discussion to get an idea of what people would want. Comments and suggestions are welcome.

KommandorKeen, 2017-07-14:
NDVI, greeness index, GLI, etc. In effect it does. Once you have an orthoimage, omatter what the channel contain then you can make an index from it. I  have done this on the dense point cloud generated it is great for pulling up different tree types. (in Cloud Compare).

juvinski, 2017-07-14:
Hi,

I'm after getting orthomosaics with nir and rgb images calculating 13 index ( ndvi, gli, esavi and several others.
There are some points, for instance, the first work is to precisely align the images, today between nir and rgb - I'm using two cameras with 8 centimeters between them, I had to do this alignment as first step. After that I'm using gdal to compare and analise the points and really getting the same coordinate point and just after that do the maths

juvinski, 2017-07-14:
first thing, would be identify the pictures as nir images and how @KommandorKeen  said, what channel are nir and what is the red channel. There are a second point, is the noise between channels, for instance, there are cameras I need to use 0.8 factor (Red channel - 0.8) and other camera 1.2 (red + 1.2).
To calculate the other index (I think we need to somehow combine both mosaics or indicate from what mosaic will be the green and blue channels. I have as I said 13 index, should I post the formulas ? All algorithms are using opencv

kikislater, 2017-07-14:
Which camera are you using Vinicius ? How do you elaborate the noise computation ? 
I have the same deal to aligning before processing. Sould be cool to have a ton a formulas but most are using GIS do to that !

juvinski, 2017-07-14:
Hi @kikislater, I was using Mapir Survey 1 and 2 models - but after a lot of problems I'm searching for a new camera.
For pictures alignment I'm using the findHomography and warpPerspective functions. All maths I'm using opencv functions and I think is very easy to develop and insert on WebODM - I was trying to do that but for me is really hard to understand this openreact :)

pierotofy, 2017-07-14:
I'm also wondering if anyone has thoughts on VARI https://support.dronedeploy.com/v1.0/docs/understanding-ndvi-data#section-visible-atmospherically-resistant-index-vari which just uses RGB cameras and how it compares with NDVI?

kikislater, 2017-07-14:
I have sequoia, rgb camera activated on sequoia and satellite datas from the same day and same hour. I could compare them. Just give me time to compare them :-/ Sequoia is weak : lack of support and poor/false informations given. They try to sell pix4d instead of proving reliable datas : they say they are making hardware not software... Ahaha WTF ! Satellite datas are corrected or providing with necessary recommandations for corections.
Sequoia need vignetting correction, then reflectance estimation ... 

First there is a difference between true ndvi and flase ndvi. False ndvi are not reliable IMO, I had one based on imx206 (same as gitup git2 mapir survey2) but stolen during travel. It's very difficult to make comparison with false ndvi in multiple time scheduled. 
I never used VARI but I will be glad to test !

juvinski, 2017-07-14:
Hi,
I just know how to calculate, I don't know what every index means. I found the formula:
https://www.harrisgeospatial.com/docs/BroadbandGreenness.html#Visible

KommandorKeen, 2017-07-14:
#kikislater sounds like you are working with many of these outputs.
The wrld is changing with respect to NDVI and the other indices. 
Satellite produced NDVI on a clear day is the benchmark in terms of reproducibility, but growers, with drones, can now collect data at the timescales that they want, My view is the single biggest improvement for interflight comparison regardless of bands used is to have an incident radiation sensor on the drone. and adjust the values of all bands accordingly  as a first step in processing. Will dig out the paper....

dakotabenjamin, 2017-07-17:
FWIW there is an open PR on ODM: https://github.com/OpenDroneMap/OpenDroneMap/pull/617

It's basically a plugin so it doesn't run by default but it could help with integration

merkato, 2017-07-18:
@pierotofy There's interesting article https://pubag.nal.usda.gov/download/59819/PDF about TGI index, VARI, and NDVI comparision in test field. I've modeled my vegetation indices plugin that Dakota means one post earlier, on that work.

abrah5283, 2018-02-27:
Hello guys, my name is Abraham Bonilla and i'd like to help integrating NDVI visualization to WebODM, I'm new at programming but i have the disposal to make that happen. I have some ideas like integrating Mapir camera control (https://github.com/mapircamera/Fiji) (https://github.com/mapircamera/MAPIR_Camera_Control) if it is possible to perform radiometric calibration and Infragram from PublicLabs (https://github.com/p-v-o-s/infragram-js) to visualize the NDVI index, i'd like to know how and where could I start. Thanks

LucasMM86, 2018-02-27:
Processing more than 3 bands together in ODM is problematic because of the alignment of the bands.
We use 2 mapir survey 2, one RGB and one NIR. 
I managed to process RGB and NIR images together and then obtain 2 aligned orthomosaics. See the atached file. Also commented here https://github.com/OpenDroneMap/OpenDroneMap/issues/500
My idea is to implement this in ODM.
[ProcessingRGBandNIR_InODM.pdf](https://github.com/OpenDroneMap/WebODM/files/1764372/ProcessingRGBandNIR_InODM.pdf)

pierotofy, 2019-01-07:
+1 for this one (just recording a user feedback).

Digdgeo, 2019-01-16:
Hi Sylvain, 

Are you able to process your SEQUOIA Tiff with OpenDrone Map? I am very interested in this task.

Thanks

> I have sequoia, rgb camera activated on sequoia and satellite datas from the same day and same hour. I could compare them. Just give me time to compare them :-/ Sequoia is weak : lack of support and [...]

dirceup, 2019-04-15:
Hi, I've made something that could be helpful to WebODM users looking to implement **VARI**, **NDVI** and others **Vegetation Indices** (VIs) on their projects.

It's published as **open source** there:
[https://github.com/dirceup/tiled-vegetation-indices](https://github.com/dirceup/tiled-vegetation-indices)

The approach is:
Generated a thumbnail of the full orthophoto. Then apply each one of the implemented VIs on the thumbnail and save a histogram of each index result.
The histogram is then used to calculate a min and max used to "clip" results bellow/above min/max on individual tiles.

I hope this contributes to the evolution of that great piece of open source software that is ODM and **WebODM**.

guilocks, 2019-09-30:
Hi guys, you managed to integrate NDVI with webODM?

Miguelistan, 2019-11-03:
Very interested on having NDVI from RGNIR cameras (like Mapir Survey 3).
The guys from Agrocam.eu are running an NDVI module, it works very well

pierotofy, 2019-11-04:
Very much on our priority list; keep an eye for changes on https://github.com/pierotofy/OpenDroneMap/tree/multispec and on this repository (I'm going to start working on a new map tiling system that includes band selection and algorithms in WebODM soon).

pierotofy, 2019-11-04:
If you have images for testing @Miguelistan, please share them with us! We're looking for more datasets (RGNIR) that we can use to evaluate results.

juvinski, 2019-11-04:
hi @pierotofy i have several datasets, how can i share with you ?

pierotofy, 2019-11-04:
If you need to keep them private you can use this form: https://masseranolabs.com/contact/ (upload to Google Drive / Dropbox, then share a link).

pierotofy, 2019-12-03:
This is now implemented as part of #746 :clinking_glasses:

closed 2019-12-03 by pierotofy