WebODM/WebODM #243: I have a task that keeps failing...

nsicart, 2017-07-15:
Hello! I have a task of 155 images that keeps failing. At the beginning it failed saying that maybe there is a lot of vegetation and it recomended to change some parameters.

I changed and now it says it run out of memory... Maybe there's something I don't know that I can change...

I share the images, thanks form helping. It's a burned area.

https://drive.google.com/open?id=0B0AuzL8kS6NdWmQ2NDdIUHJSSjg

This is the info:

https://docs.google.com/document/d/1JSAy0d4iI0KP74VY4XkSRyCCZGoi6m6aU0zOIJJpSuE/edit?usp=sharing

pierotofy, 2017-07-15:
If it says that it ran out of memory, it probably did.

Did you follow the links provided in the error message to increase the amount of memory available to docker?

nsicart, 2017-07-15:
Yes and I can't give it more RAM... My laptop has 8GB and Docker around 5,5GB... So if I give it more I don't have enough for the laptop...

pierotofy, 2017-07-15:
You might have to process your task in smaller individual tasks, decrease the size of your images, or simply fly at a higher altitude and capture fewer images covering the same area.

There's going to be memory improvements coming to opendronemap in the next few months, so I would also keep an eye out for that.

closed 2017-07-15 by pierotofy