WebODM/WebODM #244: Loading Processing Nodes

bobbyturk, 2017-07-17:
I have installed WebODM and I get this message that it is loading processing nodes. That is as far as I get. I had this installed on my mac and I didn't have this problem. I think that the processing node is supposed to load quickly.
[image]
[image]

pierotofy, 2017-07-17:
Hey @bobbyturk, yes it should load quickly.

If you expand the "Processing Node" menu on the right and click on "node-odm-1", does the processing node appear online?

Also, what happens if you try to open WebODM with a different browser (say, Firefox)? Maybe this is a browser compatibility issue.

bobbyturk, 2017-07-17:
Hi,

I switched to Firefox and it worked! Just made my first model!

thanks,

bobby

pierotofy, 2017-07-18:
Awesome!! 👍 Definitely a browser compatibility issue then. Would you mind sharing what version of Internet Explorer you were using so that we can have somebody try to fix it?

https://www.lifewire.com/what-version-of-internet-explorer-do-i-have-2617990

pierotofy, 2017-10-26:
Older browsers are now receiving a warning from WebODM to upgrade. This should be fixed.

closed 2017-10-26 by pierotofy