WebODM/WebODM #245: DSM, DTM, orthoDSM?

KommandorKeen, 2017-07-18:
You all did a great job with the DSM, works better than I thought!!

1) How do I specify that the classes calculated in the DTM processing get added to the LAS file?
2) Is it possible to create a DSM ortho photo?  Yes it would be holey, but I am trying to see the ground without the trees. Currently I take the mesh into meshlab and trim off all the above ground bits.  It works pretty well if I have captured images from multiple grid missions at  70 degrees from the vertical, not 90.

ODM/WebODM is really a contender now for commercial deployment.

ON a different note SIFT  and CUDA acceleration has been incorporated in COLMAP which claims to be open source, Are we missing something?

pierotofy, 2017-07-18:
Hey thanks 😄 

For 1) I would open an issue in the OpenDroneMap repo; this is something that should probably be added.

For 2) it would be interesting if somebody contributed a PR for this.

SIFT is patented, so while the code is open source, people would need to pay a royalty in many countries where the patent is enforced. https://www.google.com/patents/US6711293

pierotofy, 2020-04-02:
This belongs to the ODM repository; if there's still interest in adding this feature, open a PR/issue in the ODM repository. :+1:

closed 2020-04-02 by pierotofy