WebODM/WebODM #247: upload folder path change webodm webapp

boopathit98, 2017-07-19:
good afternoon sir,
sir i have doubt how to change save image folder and view map json assets url 
i have local machine process image files 
how to configuration to live server sir any possible sir

pierotofy, 2017-07-19:
Hi,

I'm not sure I understand the question. Could you try explain it in different terms?

boopathit98, 2017-07-24:
good evening sir 
i used WebODM script i will upload some image processing and how to find that images and tiles image and geotiff and obj files where it stored

pierotofy, 2017-07-24:
All results are stored in a docker volume. They are not available on the host file system. If you want to change the mapping, you need to change `docker-compose.yml` to a different path, example:

https://github.com/OpenDroneMap/WebODM/blob/master/docker-compose.yml#L21

From:
```
    volumes:
      - appmedia:/webodm/app/media
```

To:
```
    volumes:
      - /path/where/to/store/files:/webodm/app/media
```

Then restart WebODM.

closed 2017-08-18 by pierotofy