WebODM/WebODM #256: Logo, title customizations

pierotofy, 2017-07-31:
Some users have inquired about the possibility to customize WebODM (change logo, change name, etc.). This should probably be facilitated without having to change HTML/CSS, as it would encourage branding and thus wider adoption.

pierotofy, 2017-11-10:
I hereby declare this one done!

[image]

closed 2017-11-10 by pierotofy

uasimagery, 2017-11-11:
Do we just run the ./webodm.sh update command to incorporate this change? Thanks :)

Tim

[image]<https://user-images.githubusercontent.com/1951843/32680890-d718b3a4-c63a-11e7-8218-7ffb13db5367.png>

pierotofy, 2017-11-11:
Correct!

uasimagery, 2017-11-11:
Done! looks great :) Awesome Job!

Silly question: will the update command update all changes that are made to the WebODM software, even if it isn’t directly related to the webodm.sh file?

Just wanting to stay up to date thats all…..

Best regards,
Reception Desk
Administration

[https://ucarecdn.com/f075f5ce-20c8-4430-a6e3-68edce6b795f/-/resize/166x73/img.png]<http://www.uasimagery.com.au/>
p.              1800 UAS IMG
m.              +61 423 757 851
w.              www.uasimagery.com.au<https://www.uasimagery.com.au/>
a.              Unit 2, 767 Sandgate Rd
Clayfield, QLD 4011

[https://app.zippysig.com/social-icons/24/000055/06facebook.gif]<http://facebook.com/uasimagery>  [https://app.zippysig.com/social-icons/24/000055/06twitter.gif] <http://twitter.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06instagram.gif] <http://instagram.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06youtube.gif] <https://www.youtube.com/channel/UCMxm268BPm1l_vgo12Ey5oA>   [https://app.zippysig.com/social-icons/24/000055/06linkedin.gif] <http://www.linkedin.com/uas-imagery>

pierotofy, 2017-11-11:
Thank you!

I’m not sure I understand the question. What do you mean by “even if it isn’t directly related to the WebODM software”? The update command should take care of everything.

uasimagery, 2017-11-11:
So for example, if you made an amendment to webpack.config.js<https://github.com/OpenDroneMap/WebODM/blob/master/webpack.config.js>, would "./webodm.sh update” update the javascript file too?

Best regards,
Reception Desk
Administration

[https://ucarecdn.com/f075f5ce-20c8-4430-a6e3-68edce6b795f/-/resize/166x73/img.png]<http://www.uasimagery.com.au/>
p.              1800 UAS IMG
m.              +61 423 757 851
w.              www.uasimagery.com.au<https://www.uasimagery.com.au/>
a.              Unit 2, 767 Sandgate Rd
Clayfield, QLD 4011

[https://app.zippysig.com/social-icons/24/000055/06facebook.gif]<http://facebook.com/uasimagery>  [https://app.zippysig.com/social-icons/24/000055/06twitter.gif] <http://twitter.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06instagram.gif] <http://instagram.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06youtube.gif] <https://www.youtube.com/channel/UCMxm268BPm1l_vgo12Ey5oA>   [https://app.zippysig.com/social-icons/24/000055/06linkedin.gif] <http://www.linkedin.com/uas-imagery>

pierotofy, 2017-11-11:
It will attempt to merge any change that is conflicting between files, yes. But sometimes it won't be able to and you'll get an error and will have to resolve the conflict manually.

uasimagery, 2017-11-11:
Ok great. Thanks so much for your help. So what process do I follow to update files outside of WebODM.sh?

Best regards,
Reception Desk
Administration

[https://ucarecdn.com/f075f5ce-20c8-4430-a6e3-68edce6b795f/-/resize/166x73/img.png]<http://www.uasimagery.com.au/>
p.              1800 UAS IMG
m.              +61 423 757 851
w.              www.uasimagery.com.au<https://www.uasimagery.com.au/>
a.              Unit 2, 767 Sandgate Rd
Clayfield, QLD 4011

[https://app.zippysig.com/social-icons/24/000055/06facebook.gif]<http://facebook.com/uasimagery>  [https://app.zippysig.com/social-icons/24/000055/06twitter.gif] <http://twitter.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06instagram.gif] <http://instagram.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06youtube.gif] <https://www.youtube.com/channel/UCMxm268BPm1l_vgo12Ey5oA>   [https://app.zippysig.com/social-icons/24/000055/06linkedin.gif] <http://www.linkedin.com/uas-imagery>

pierotofy, 2017-11-11:
Depends on what files you modified, and if there are conflicts. Are there?

uasimagery, 2017-11-11:
I haven’t updated anything other than the WebODM.sh file we discussed... just wondering for future reference that’s all.

Best regards,
Reception Desk
Administration

[https://ucarecdn.com/f075f5ce-20c8-4430-a6e3-68edce6b795f/-/resize/166x73/img.png]<http://www.uasimagery.com.au/>
p.              1800 UAS IMG
m.              +61 423 757 851
w.              www.uasimagery.com.au<https://www.uasimagery.com.au/>
a.              Unit 2, 767 Sandgate Rd
Clayfield, QLD 4011

[https://app.zippysig.com/social-icons/24/000055/06facebook.gif]<http://facebook.com/uasimagery>  [https://app.zippysig.com/social-icons/24/000055/06twitter.gif] <http://twitter.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06instagram.gif] <http://instagram.com/uasimagery>   [https://app.zippysig.com/social-icons/24/000055/06youtube.gif] <https://www.youtube.com/channel/UCMxm268BPm1l_vgo12Ey5oA>   [https://app.zippysig.com/social-icons/24/000055/06linkedin.gif] <http://www.linkedin.com/uas-imagery>

pierotofy, 2017-11-11:
If that happens in the future, follow this guide: https://githowto.com/resolving_conflicts 😄