WebODM/WebODM #257: Create dynamically html from jsx files

jhouedanou, 2017-08-02:
Hi and thanks for this fantastic software.
I am currently working on a template for WebODM which requires me to modify some of the jsx. 
I know there is a task runner inside of webodm - webpack - but i dont know how to run it from the command line.
Could you point me in the right direction.
Thank you.

closed 2017-08-03 by jhouedanou

pierotofy, 2017-08-03:
Hi! Does this link answer your question? http://docs.webodm.org/#docker-setup you can use the devenv.sh script to automatically watch for jsx changes.

Also would love to hear more about the customizations you are trying to make, perhaps they could be included in the webodm codebase if you are interested in sharing them.

jhouedanou, 2017-08-03:
Hi, that did the trick, thanks.
As fas customizations are concerned , I am currently removing bootstrap in favor of Materialize.css - http://materializecss.com/ and also adding some other fancy stuff . 
They are not finished yet , but I'll let you know as soon as they are and share the repository with you