WebODM/WebODM #258: WebODM Paralyzed execution

stiward, 2017-08-06:
Hello dears,

I am quite new to the world of Linux and VM, to start the code the file "webodm.sh" the whole process started well (after several things that solved themselves, I have no idea how to step), only gave an error at the beginning (Image 1) but then continue. But, I have a problem, the terminal has more than half an hour that does not move from this line: webapp | [2017-08-05 23:07:19 +0000] [51] [INFO] Booting worker with pid: 51. I will attach the image so that they can see.

Please let me know if there is anything wrong or how to proceed.

Thank you!

Image 1
[image]

Image 2
[image]

pierotofy, 2017-08-06:
Looks like WebODM is up and running, but for some reason you didn't receive the "success" message that would have told you to open a browser to http://localhost:8000 and log-in.

I'll leave this open as a bug, not sure why that happened.

What distro are you using?

stiward, 2017-08-06:
I'm using Docker Quicktime Terminal in Microsoft 10

stiward, 2017-08-06:
It's strange, I'm coming in. I'll tell you if there is any error in processing the images.

pierotofy, 2020-04-02:
Lots of improvements have been added since 3 years ago; if this is still happening, please re-open. :clinking_glasses:

closed 2020-04-02 by pierotofy