WebODM/WebODM #260: Register user via API

hy9be, 2017-08-11:
Hi, do we have an API to register user? Like ``/api/user/new`` Thanks!

pierotofy, 2017-08-12:
Hi, currently we do not. This is something that should be added. I'd be happy to guide volunteers into implementing this one.

hy9be, 2017-08-13:
@pierotofy Please give me some directions, I will gave it a try.

pierotofy, 2017-08-15:
1. You'll want to get familiar with setting up a development environment if you haven't already: http://docs.webodm.org/#for-developers

2. WebODM uses Django and Django REST Framework to handle the the API endpoints: http://www.django-rest-framework.org/ You don't necessarily need to understand all of the frameworks' functionalities, but you should at least be familiar with the concepts of ModelSerializer http://www.django-rest-framework.org/api-guide/serializers/#modelserializer and authentication via https://django-guardian.readthedocs.io/en/stable/.

3. Good examples to read from the WebODM codebase are probably in `/app/api/processingnodes.py`. We would be adding API endpoints for the django.contrib.auth.models.User model (https://docs.djangoproject.com/en/1.11/topics/auth/default/#user-objects) instead of ProcessingNode though.

4. Unit tests are a must; see `app/tests/test_api.py` for examples on how to test the API to verify it's working appropriately.

This is just an overview, if you need more information please ask. 👍

emialonzo, 2019-09-23:
And this is not working? https://docs.webodm.org/#admin-users

> POST /api/admin/users/{id}/

Im getting always 404 error.

pierotofy, 2019-09-23:
@w-toguchi83 might know the answer, but we should move questions to the forum. https://community.opendronemap.org

w-toguchi83, 2019-09-24:
@emialonzo (cc: @pierotofy )

I'm sorry. There is a mistake in the documentation. (here: https://docs.webodm.org/#update-a-user)

(mistake) `POST /api/admin/users/{id}/`

(right) `PUT /api/admin/users/{id}/`

I will correct the documentation soon.

See also the following test code:
https://github.com/OpenDroneMap/WebODM/blob/master/app/tests/test_api_admin.py

If you want to create a new user, please make a request to this endpoint.

`POST /api/admin/users/`

(https://docs.webodm.org/#create-a-user)

pierotofy, 2019-11-07:
This has been implemented a while ago: https://docs.webodm.org/#create-a-user

closed 2019-11-07 by pierotofy