WebODM/WebODM #261: Crashing on georeferencing

AnthonyCoughlin, 2017-08-12:
Could you guys give me some guidance on this.  Trying to build a 3D model of a monument.  Receiving an error when the georeferencing model script runs.  The exif data holds the gps data in decimal degrees, does it need to be in dd mm ss?

Thanks for the help.

[task_output _VetMem.txt](https://github.com/OpenDroneMap/WebODM/files/1219370/task_output._VetMem.txt)

pierotofy, 2017-08-12:
Hey @AnthonyCoughlin, this is part of the output from an image known to work:

`exiftool DJI_0018.JPG | grep GPS`

```
GPS Altitude                    : 198.3 m Above Sea Level
GPS Latitude                    : 46 deg 50' 33.39" N
GPS Longitude                   : 91 deg 59' 40.42" W
```

So perhaps compare it with your input dataset.

AnthonyCoughlin, 2017-08-12:
Where can I find the exif tool that you used?

kikislater, 2017-08-12:
Here : https://sno.phy.queensu.ca/~phil/exiftool/
And in your linux distribution repository

AnthonyCoughlin, 2017-08-13:
Thanks for the help thus far.  I've downloaded version 10.60 and extracted the tool to a folder I created (EXIFTOOL) in my WINDOWS directory.  I have tried examining several images and each time the tool's cmd prompt flashes and I can see some data for an instant and it immediately closes.  I've been able to stop it from closing by clicking on it but it only gives me a partial list of data.  I've attached one of the images, would one of you guys mind testing as well?  I downloaded an image from ODM\benchmark and experienced the same result.  I've also downloaded version 10.55 and am experiencing the same issue.  I'm going to download a some other versions and see if  I can get one to work.

Again, thanks for the help!

[image]

AnthonyCoughlin, 2017-08-13:
Update: i should have mention this at the beginning, I'm using Win10 64 bit.  I've tried 10.60, 10.59, and 10.55. All with the same results command prompt flashes with data and then immediately closes.

pierotofy, 2017-08-13:
@AnthonyCoughlin, see this link: https://www.computerhope.com/issues/ch000738.htm

AnthonyCoughlin, 2017-08-13:
@pierotofy 

Once again Sir, you are the man!

AnthonyCoughlin, 2017-08-15:
I also noticed that regarding the parameters you listed in your example above, they are listed twice in my EXIF data (see attached).

Thanks for the help.

[Xstar_EXIF_LatLongAlt.txt](https://github.com/OpenDroneMap/WebODM/files/1225474/Xstar_EXIF_LatLongAlt.txt)

pierotofy, 2017-08-18:
Could you upload a few images (20ish) from your dataset with the new exif tags for somebody to try to process?

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/crashing-on-georeferencing/46

closed 2017-10-19 by pierotofy