WebODM/WebODM #262: Divide by zero error.

blemmer, 2017-08-13:
Hello

I'm new to WebODM and it seems like an awsome project :) But when I'm running WebODM with my own phantom 3 test pictures i get an error. I'm using the WebODM docker image. 

Thank you very much.

My pictures are here:
https://www.dropbox.com/sh/eik57yy118t8qch/AADeErAijoeOIn-HgTuwb2kVa?dl=0

And the Error from the log is here:
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /var/www/data/2cec0fd0-7c2f-4881-802f-3190622777ec/opensfm

Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/extract_metadata.py", line 29, in run
    d = exif.extract_exif_from_file(data.load_image(image))
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 83, in extract_exif_from_file
    d = exif_data.extract_exif()
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 247, in extract_exif
    focal_35, focal_ratio = self.extract_focal()
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 166, in extract_focal
    get_frac_tag(self.tags, 'EXIF FocalLength'),
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 34, in get_frac_tag
    return eval_frac(tags[key].values[0])
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 14, in eval_frac
    return float(value.num) / float(value.den)
ZeroDivisionError: float division by zero
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 85, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

pierotofy, 2017-08-13:
Hi, welcome!

It looks like your images are missing the EXIF "Focal Length" field information. I think a more recent version of opensfm (not included in WebODM yet) handles the case where this field is missing, but for the moment the workaround is to either set the field manually (via exiftool, look up on Google where to find the program) or to pass the `--force-focal` parameter via the WebODM interface (change the Default options to include the appropriate value when creating a new task). I'm not 100% sure the second method will work, but it's the quickest to try.

[image]

blemmer, 2017-08-13:
Thank you very much for the fast reply.

I tried your second method, but it didn't work. I will try to set the parameter manually in the metadata :)

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/divide-by-zero-error/45

closed 2017-10-19 by pierotofy