WebODM/WebODM #263: ViewMap preview zoom-in

keum, 2017-08-15:
Hello, 
I've used WebODM to create orthomoasic photo using gps camera and long pole.  I created very small area. I was able to preview 2D map nicely except I wanted to zoom in closer but stops zoom at certain level at google map basemap. was wondering if i can zoom in closer. thanks

pierotofy, 2017-08-15:
Hi,

We currently set the map levels limit to `21`. You would need to modify both node-opendronemap (https://github.com/OpenDroneMap/node-OpenDroneMap/blob/master/scripts/postprocess.sh#L41) and WebODM to enable a higher level (https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/components/Map.jsx#L178). We might change this in the future to allow for custom zoom levels. Also consider simply downloading the resulting orthophoto and opening it in QGIS or some other GIS program.

keum, 2017-08-15:
Thank you for your reply, suggestion to view through QGIS is what I'll do for now and will serve my need for sure. thanks! great work and love the software!

pierotofy, 2020-04-02:
We now have dynamic tiles and you can zoom-in 'til you can see individual pixels! :clinking_glasses:

closed 2020-04-02 by pierotofy

abrichr, 2020-09-25:
@pierotofy can you please clarify how this feature can be enabled? I cloned the repo yesterday and am deploying the master branch using Docker. I can see the orthophoto on the base map, but it disappears once I get past a certain zoom level.

Maximum zoom level showing orthophoto:

[image]

One zoom level further:

[image]

The NodeODM API version is 2.0.0. Please let me know if I can provide any additional information. Thanks!