WebODM/WebODM #264: Bellus Example

kirby031, 2017-08-17:
Hello i had a problem with the bellus example, first i ran it by default and it didn´t work, then i tried running with a high SFM number of processes, and i enabled the option "use-exif", but it return me a bad georreferencing Orthophoto; it was in the place but the image was smaller than te real map world. So my questions are:
* How does SFM number of processes work?
* What does "use-exif" option do? if i didn´t enabled the option, it doesn´t work?
* And what happened with the georreferencing?
* When you download bellus example, it has a ".gitignore" file, do i have to put this in folder "WebODM\app\migrations" ? or why does the example gives this file?

I am running WebODM in windows 10 with a docker container; my laptop has 8GB Ram; i ran the oracle virtual machine with 5 RAM and 2 processor.
Thanks a lot

pierotofy, 2017-08-17:
- Higher sfm processes speed up the computation but use more memory.
- It ignores the ground control points file (if one is provided) and uses the geolocation information from EXIF tags (embedded in the images)
- I'm not sure; the Bellus dataset is quite old, I would recommend using a more recent dataset for performing tests and see if the error persists.
- .gitignore is an internal file for version control. You do not need to include it anywhere. Just upload the .JPG and the .txt (GCP) file.

Hope this helps.

kirby031, 2017-08-17:
Yes, thanks a lot. I have two more questions.
* how can i know the ideal sfm processes  number to input?
* And if i don´t enable the "use-exif" option, webODM work with "GCP" file and "EXIF" data? or wich one is better to use?

dakotabenjamin, 2017-08-17:
ODM automatically detects a GCP file if it exists but if not it will default to EXIF. The use-exif tag is only really useful for if you have GCP but want to override and use EXIF for georeferencing.

kirby031, 2017-08-17:
Yes, i tried running it again with ubuntu but without checking "use-exif" and it gives me the same solution as when i ran it with windows. Why does it doesn´t get the correct georreferencing?

An Image:
[image]

Thanks a lot

dakotabenjamin, 2017-08-17:
Try enabling skip-resize

kirby031, 2017-08-17:
Thanks, i´ĺl try. What does "skip-resize" do?

dakotabenjamin, 2017-08-17:
So I'm not in the office right now to confirm the issue but I believe that because the gcp was made for the full size image, when you resize the pixels won't line up.

kirby031, 2017-08-18:
Thanks a lot dakotabenjamin, it worked and ran in half time. So when i ran the images by default it resizes? is The resize problem common?

dakotabenjamin, 2017-08-18:
Only when working with GCPs. It's a bit of archaic code that now is not so relevant. I'm going to try to spend some time this week removing the whole process.

kirby031, 2017-08-18:
are you trying to say that GCP alredy isn´t used?

dakotabenjamin, 2017-08-18:
Sorry. I meant the Resize module.

kirby031, 2017-08-18:
Oh, Thanks a lot for the support :)