WebODM/WebODM #266: Post Flight Checker

wildintellect, 2017-08-18:
Process for reviewing data collected while in the field to make sure you captured enough data and overlap.
Similar to Stephen's sfcgal tool on his blog, but simpler:

* Plot the Exif locations.
* Show the direction and speed of aircraft (points colored by speed.
* Based on the height, camera parameters, and orientation estimate the footprint of the images.
* Calculate the overlap between photos (perhaps a heatmap style raster of # of images per pixel covered.

kikislater, 2017-08-18:
Formula for footprint is here (link to github in video description) :
https://youtu.be/aZP1bBvMXSI

pierotofy, 2017-08-18:
Agree this would be super useful. Is there code available somewhere?

kikislater, 2017-08-18:
Yes in this commit : https://github.com/DroidPlanner/Tower/commit/b5173de4261aa6ca0cec105e0d3107e5bab7eaff#diff-756759bedaecad14afebba69b81639f7

I'm not kind fan of java and I still have to decode some parts but it's documented. It seems need some steps : 

- This one is easy : 3D rotation matrix from vehicule IMU (so flight log is needed). Document is not present at this url but with google we could find it
	 * Create a rotation matrix given some euler angles
	 * this is based on http://gentlenav.googlecode.com/files/EulerAngles.pdf

It generates a 3x3 rotation matrix. Here is the code : https://github.com/DroidPlanner/Tower/blob/b5173de4261aa6ca0cec105e0d3107e5bab7eaff/Core/src/org/droidplanner/core/helpers/math/MathUtil.java

- Then GSD and flat footprint from camera is needed. Standard flight planning computation. Watch here : https://github.com/DroidPlanner/Tower/blob/b5173de4261aa6ca0cec105e0d3107e5bab7eaff/Core/src/org/droidplanner/core/survey/Footprint.java

- From these two previous series of value, they compute a new center coordinates for picture
	**
	 * based on http://www.asprs.org/a/publications/pers/2005journal/july/2005_july_863-871.pdf 
	 *
- Next they compute lateral and longitutinal size from java gis module = geotools : here I'm lost with geotools.

With new coordinates, lateral/longitudinal size you have a new polygon

smathermather, 2017-08-18:
I've got an alternate approach documented in the PostGIS Cookbook which leverages SFCGAL in Postgres. It would need some additional refinement and generalization, but if it would be useful, I can carve out some time to implement and extend.

kikislater, 2017-08-19:
Hum interesting, got it ! Chapter 7, next time I will go to foss4g south africa ...
Amazing, thank you for sharing. I have to try it. ST_RotateX and ST_RotateY seems good now.

pierotofy, 2020-04-02:
This is implemented in https://github.com/uav4geo/DroneDB/ (although it's not friendly, yet)

pierotofy, 2021-10-14:
This is definitely under the scope of DroneDB (and DroneDB Desktop) at the moment. Not sure there's an immediate need in WebODM.

https://github.com/DroneDB/ddb-desktop

Closing.

closed 2021-10-14 by pierotofy