WebODM/WebODM #269: Problem Running

kirby031, 2017-08-21:
Hello, I am trying to run this 2 examples and In both the same thing happens, it freeze when is "computing depthmap for image.... ". I am running in Ubuntu 16.04 RAM 8GB Video 940M core i5; i couldn´t take a screenshot because my ubuntu doesn´t do anything, it freezes.
Thanks a lot.
http://www.tuffwing.com/support/ReachGPS_Pix4D_sample_data.html
https://support.pix4d.com/hc/en-us/articles/202561419-Example-Datasets-Available-for-Download-Quarry#gsc.tab=0

pierotofy, 2017-08-21:
Try to reduce the number of the "opensfm-processes" option from the task options when you create one. This is a known bug that will be fixed as soon as the latest changes of opensfm are integrated.

Thanks for reporting the problem!

kirby031, 2017-08-22:
Oh thanks, is it the same case for when i am running images and it shows that it is "merging", but it never ends, i update the page and it returned to 1:25 hours when i was around 5 hours.
Thanks a lot

pierotofy, 2017-08-22:
Most likely, yes. See also https://github.com/mapillary/OpenSfM/issues/190

kirby031, 2017-08-22:
Thanks, so what number of opensfm-processes do you recommend me? or should i use pvms?

pierotofy, 2017-08-22:
pvms is a good option that will yield a worse point cloud, but you should still get a good orthophoto (and does not suffer from the memory problem). I usually set opensfm-processes to `1` to start with, and progressively increase it (depending on the number of input images).

smathermather, 2017-08-23:
We have a fix in the pipeline for this -- we just need to upgrade to the latest OpenSfM master, which fixes the out of memory problem with merging. I think @dakotabenjamin was going to look into this later this week.

kirby031, 2017-08-23:
Oh thanks a lot for the support. I´ll try what @pierotofy recommends to me and @smathermather i´ll wait for the upgrade.
Thanks a lot

pierotofy, 2017-10-26:
This should be fixed. @kirby031 please let me know if the problem appears again.

closed 2017-10-26 by pierotofy