WebODM/WebODM #274: No usable processing nodes.

uasimagery, 2017-08-26:
Hi All,

I'm hoping someone can point me in the right direction. I have OpenDroneMap working with Docker and everything goes well until I upload images for processing. It seems there are no usable processing nodes. I have changed permissions for the "admin" user but this hasn't seemed to have changed anything. Is there any chance some one could walk me through fixing this issue? I have provided some screenshots which show current settings.... System specs are below in case that all helps. Any help would be much appreciated.....

Windows 10 Home
Intel Core i7-4790 CPU @ 3.60 GHz
32.0GB RAM
64 Bit OS x64 based processor

VirtualBox 5.1.26 r117224(Qt5.6.2)
Docker Version 17.06.0-ce build 02c1d87
[image]
[image]
[image]
[image]

pierotofy, 2017-08-29:
Hi, sorry you're having issues.

Could you try to delete the processing node from the admin interface and restart webodm? It should be recreated automatically when you restart (does it appear in the menu on the right under "Processing nodes"? Also did this problem occur from a fresh install, or did you change some settings before it stopped working?

uasimagery, 2017-08-30:
Thanks for the advice, I ended up having to uninstall and reinstall docker, VB and then re-clone from Github. All seems to be working now :) 

Is there a facility to add nodes yet?

pierotofy, 2017-08-30:
Yay, glad it worked!

Could you expand on the "is there a facility" question? I'm not sure I understand it fully.

uasimagery, 2017-08-31:
Sure, as I understand it, it might be possible to run multiple nodes at once while processing several jobs at on time... is this the case?

pierotofy, 2017-08-31:
Yes adding more processing nodes will allow you to run multiple jobs in parallel. See https://github.com/OpenDroneMap/node-OpenDroneMap on how to setup new nodes.

closed 2017-09-01 by pierotofy