WebODM/WebODM #275: Booting worker with pid: 51

faraztehrani, 2017-08-29:
Hello,

I followed the instruction on https://github.com/OpenDroneMap/WebODM/blob/master/README.md to install the program using Docker. At some point during installation it stopped further progress. Here is the last line that it printed on screen: 
Booting worker with pid: 51

It seems it has something to do with Gunicorn.

Do you know how I should proceed?

Thanks,

pierotofy, 2017-08-29:
Looks like a duplicate of #258 

Are you on Windows?

faraztehrani, 2017-08-29:
Yes. The same problem. But, I didn't find a solution over there.

I use Windows 7.

pierotofy, 2017-08-29:
Run `docker-machine ip` from the terminal. Use that value (the IP address, which looks like x.x.x.x) to open a web browser to http://x.x.x.x:8000.

faraztehrani, 2017-08-29:
Great! 

Thanks for your help.

closed 2017-08-29 by faraztehrani