WebODM/WebODM #276: Geo-referencing Inaccurate

dsandson, 2017-08-29:
I've been trying out WebODM for a little while now, and trying to gain enough knowledge to incorporate it into my work.

I have installed Ubuntu 16.04 LTS, and got WebODM up and running in Docker.

I gathered a data set with RTK GPS surveyed GCPs, then used the recommended technique of loading the images in QGIS & building a GCP list.txt file, with coords set to EPSG:29902 (Irish Grid). I also realise that georeferencing has issues with the default image resizing, so I've ran the process with 'skip-resize'.

When the process has finished I have loaded the original GCPs in QGIS, alongside the orthophoto, both set to EPSG:29902. However I would expect the GCPs visible on the ortho to align with the point layer GCPs. However they are out by between 8.9-12.4 meters.

The process was run with the following flags;
min-num-features: 8000, mesh-size: 150000, dem-terrain-type: ComplexNonForest, skip-resize: true, use-25dmesh: true

I've checked the gcp file, with the original survey coords and they're correct. also double checked the image coords. Cant see what else is incorrect. 

I've used the following guidance on compiling the GCP file;
https://smathermather.com/2015/03/01/opendronemap-gcp-file-format/
Not sure if this is up to date, or if theres a new improved technique.

Screenshot from QGIS and WebODM log (hopefully) attached, as well as GCP file. I can provide a link to the raw images if needed.

Regards
David

dsandson, 2017-08-29:
Sorry, uploads didnt attach to files linked below
https://drive.google.com/file/d/0B7_UoiZt_fsydzZULVVnZGJvR0k/view?usp=sharing
https://drive.google.com/file/d/0B7_UoiZt_fsySDY4NnJyMzZXU2c/view?usp=sharing
https://drive.google.com/file/d/0B7_UoiZt_fsyaDhPdDl5QTlpTG8/view?usp=sharing

pierotofy, 2017-08-29:
Mm, this could be a software fault. What happens if you use UTM coordinates for the GCP file? The fact that the georeferencing is off by a few meters might be a problem related to reprojection.

dsandson, 2017-08-29:
Thank you Piero,

I've loaded another orthophoto produced using just the exif data from the onboard gps, and they're about 0.5m out as I'd expect.

We don't use UTM often, but I'll convert the coords to ED50 UTM 30N in qgis in the morning and see How it works out.

dsandson, 2017-08-30:
Well, 

I converted the GCP coords to ED50 UTM 30N EPSG:23030 in QGIS, and with reprojection on-the-fly enabled it showed the new point layer as perfectly overlapping the existing points in Irish Grid (29902).

Transferred the new coords into the GCP_list.txt file and amended the EPSG code, and the output still exhibits the same X&Y error once imported into QGIS, with both layers loaded as EPSG 23030.

Photos and associated GCP files are linked below;
https://drive.google.com/file/d/0B7_UoiZt_fsyVE9XLUFVMDNScFk/view?usp=sharing

Regards
David

pierotofy, 2017-08-30:
Thank you for trying. If my analysis is correct, I think the problem is instead related to the datum used (in this case, ED50). I don't think OpenDroneMap currently handles well datum different than WGS84. 

https://github.com/OpenDroneMap/OpenDroneMap/blob/ea5aeef213354a56e2c0a4296e236c4c53885161/modules/odm_extract_utm/src/UtmExtractor.cpp#L219

https://github.com/OpenDroneMap/OpenDroneMap/blob/1385ab12168adc6df34b5c2f274dbc3a78d186d3/opendm/types.py#L290

@dakotabenjamin can you confirm this? If so I'll go ahead and open an issue under the ODM repository.

dakotabenjamin, 2017-08-30:
If they used the EPSG code then it shouldn't be an issue.  https://github.com/OpenDroneMap/OpenDroneMap/blob/1385ab12168adc6df34b5c2f274dbc3a78d186d3/opendm/types.py#L300

For something to be a few meters off maybe it's a floating point precision issue from the conversion?

dakotabenjamin, 2017-08-30:
@dsandson you may want to try to calibarate your images. Distortion from each image will propagate in error in the orthophoto which might lead to further issues. I see issues like this a lot with DJI cameras especially. 

See: https://github.com/dakotabenjamin/cameracalibration

dsandson, 2017-08-30:
dakotabenjamin thank you, I'll give that a go. Looks like that might be a useful addition to ODM/WebODM too.

I'd be surprised if that was the only issue. As I said using only the WGS84 coords from the EXIF data produces errors in the range of 0.5m. 

This may take a few days to get done.

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/geo-referencing-inaccurate/44

closed 2017-10-19 by pierotofy