WebODM/WebODM #278: Cannot change where files are stored

KommandorKeen, 2017-08-31:
Followed the instructions (I think) and modified my docker-compose.yml  to point to a different point in my native (LINUX) file system, but nothing is going there and my docker used space on my system drive (which isn't too big) keeps growing.

I am really missing some basic concept of dockers and filesystems so please help.
docker.compose.yml looks like this:

version: '2'
volumes:
  dbdata:
    driver: local
  appmedia:
    driver: local
services:
  db:
    image: opendronemap/webodm_db
    container_name: db
    ports:
      - "5432"
    volumes:
      - dbdata:/var/lib/postgresql/data
  webapp:
    image: opendronemap/webodm_webapp
    container_name: webapp
    entrypoint: /bin/bash -c "chmod +x /webodm/*.sh && /bin/bash -c \"/webodm/wait-for-it.sh db:5432 -- /webodm/start.sh\""
    volumes:
      - /home/simon/2TB/WebODM:/webodm/app/media
    ports:
      - "8000:8000"
    depends_on:
      - db
    restart: on-failure:10

pierotofy, 2017-08-31:
What happens if you run:

```
#./webodm.sh down
#./webodm.sh start
```

?

Does it work then?

KommandorKeen, 2017-08-31:
Nope, 
Happily continues to fill my system disk inside docker containers.
Which is why I am doing the processing using ODM direct that is compiled locally. Shame really as the web interface is nice, and getting better.  Earlier I did set up[ my containers on a different hard disk, but that hard disk failed and tbh that didn't solve the problem of access to the intermediate files for problem solving.

simon@T3600:~$ cd ~/2TB/WebODM
simon@T3600:~/2TB/WebODM$ ./webodm.sh down
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Tearing down WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml down
Stopping webapp ... done
Removing webapp ... done
Removing node-odm-1 ... done
Removing db ... done
Removing network webodm_default
simon@T3600:~/2TB/WebODM$ ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting db ... done
Starting node-odm-1 ... done
Starting webapp ... done
ERROR: No containers to start
Creating network "webodm_default" with the default driver
Creating node-odm-1
Creating db
Creating webapp
Attaching to db, node-odm-1, webapp
webapp        | wait-for-it.sh: waiting 15 seconds for db:5432
db            | LOG:  database system was interrupted; last known up at 2017-08-31 21:42:24 UTC
db            | LOG:  database system was not properly shut down; automatic recovery in progress
db            | LOG:  redo starts at 0/D6315ED0
db            | LOG:  invalid record length at 0/D631F610
db            | LOG:  redo done at 0/D631F5E8
db            | LOG:  last completed transaction was at log time 2017-08-31 21:44:41.041172+00
db            | LOG:  MultiXact member wraparound protections are now enabled
db            | LOG:  database system is ready to accept connections
db            | LOG:  autovacuum launcher started
db            | LOG:  incomplete startup packet
webapp        | wait-for-it.sh: db:5432 is available after 2 seconds
webapp        | 
webapp        |  _       __     __    ____  ____  __  ___
webapp        | | |     / /__  / /_  / __ \/ __ \/  |/  /
webapp        | | | /| / / _ \/ __ \/ / / / / / / /|_/ / 
webapp        | | |/ |/ /  __/ /_/ / /_/ / /_/ / /  / /  
webapp        | |__/|__/\___/_.___/\____/_____/_/  /_/   
webapp        | 
webapp        | 
webapp        | Checking python version... 3.x, good!
webapp        | Checking GDAL version... GDAL 2.2.1, released 2017/06/23, excellent!
webapp        | Running migrations
webapp        | Generated secret key
webapp        | INFO Booting...
webapp        | WARNING Debug mode is ON (for development this is OK)
webapp        | INFO Starting background scheduler...
webapp        | Operations to perform:
webapp        |   Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
webapp        | Running migrations:
webapp        |   No migrations to apply.
node-odm-1    | info: No tasks dump found
node-odm-1    | info: Checking for orphaned directories to be removed...
node-odm-1    | info: Server has started on port 3000
webapp        | [2017-08-31 22:16:21 +0000] [45] [INFO] Starting gunicorn 19.7.1
webapp        | [2017-08-31 22:16:21 +0000] [45] [INFO] Listening at: unix:/tmp/gunicorn.sock (45)
webapp        | [2017-08-31 22:16:21 +0000] [45] [INFO] Using worker: sync
webapp        | [2017-08-31 22:16:21 +0000] [51] [INFO] Booting worker with pid: 51
webapp        | 
webapp        | 
webapp        | Congratulations! └@(･◡･)@┐
webapp        | ==========================
webapp        | 
webapp        | If there are no errors, WebODM should be up and running!
webapp        | 
webapp        | Open a web browser and navigate to http://localhost:8000
webapp        | 
webapp        | NOTE: Windows users using docker should replace localhost with the IP of their docker machine's IP. To find what that is, run: docker-machine ip
webapp        | INFO Booting...
webapp        | WARNING Debug mode is ON (for development this is OK)
webapp        | INFO Starting background scheduler...
webapp        | WARNING Session data corrupted
webapp        | WARNING Session data corrupted
webapp        | WARNING Session data corrupted
webapp        | INFO Acquiring lock: Task [Task of 2017-08-31T22:19:05.272Z] (16)
webapp        | INFO Processing... Task [Task of 2017-08-31T22:19:05.272Z] (16)
node-odm-1    | info: About to run: python /code/run.py --resize-to 1000 --verbose --use-25dmesh --opensfm-processes 6 --project-path /var/www/data --pmvs-num-cores 12 b27e4c9f-388b-4a8d-949c-6b547298208b

pierotofy, 2017-08-31:
Just to double check, is `/home/simon/2TB/WebODM` an empty directory?

KommandorKeen, 2017-08-31:
I have tried it empty 
I have tried it with the /webodm/app/media directory structure in place.
neither works

KommandorKeen, 2017-08-31:
Just did a ./webodm.sh rebuild and confirmed the directory was empty
It worked, thanks.

closed 2017-08-31 by KommandorKeen