WebODM/WebODM #282: RTK & PPK

uasimagery, 2017-09-06:
Hi Everyone,

Total noob here but loving WebODM! Just wondering how WebODM works with PPK? could someone please shed some light on how the workflow would work? Thanks so much!

Tim

pierotofy, 2017-09-06:
We don't currently have built-in support for either RTK or PPK. Only GCPs. This is definitely something that we should add.

dakotabenjamin, 2017-09-06:
We have an example dataset that was taken with RTK, but the locations were just hardcoded into the EXIF data. What kind of support is needed to allow RTK?

dakotabenjamin, 2017-09-06:
https://github.com/OpenDroneMap/odm_data_waterbury

kikislater, 2017-09-06:
GPS coordinates in exif are normally less precise than a text file.

KommandorKeen, 2017-09-11:
I have been inserting PPK positions into my DJI Phantom 4 images. This has reduced my position uncertainty  in the georeferencing log from 5+ metres to sub metre. I am not sure how to bring it down anymore. In absolute terms this is seen as 2.5m shift to the north in the unPPK images and around a 1m shift in the PPK inserted images.  
Have yet to turn (back) on the Use position part of openSFM in the open sfm config if this is the way to do it still.

uasimagery, 2017-09-11:
Any chance you could shed some light on how I would go about doing this also Kommandor?

KommandorKeen, 2017-09-12:
Hi Tim,
I am in Hobart, so just give me a call on 0438086322 and we can discuss.
Simon

dakotabenjamin, 2017-09-12:
@kikislater is it a product of the coords themselves being recorded by a mediocre gps unit or a software issue? If I'm manually writing the camera positions by RTK I can't see how that would be different in ODM than if I supplied a text file.

kikislater, 2017-09-12:
Hi Benjamin,

Let me took an example of your ebee photo : 
```
$ exiftool DSC07492.JPG | grep 'GPS Position'
GPS Position                    : 44 deg 20' 26.63" N, 72 deg 44' 57.58" W
```

So format is like that : 
0° 00′ 0.36″

As I don't want to explain projection here, took this link to check WGS84 precision : 
https://en.wikipedia.org/wiki/Decimal_degrees

I this example we have this : 
decimal places | decimal degrees | DMS | qualitative scale that can be identified | N/S or E/W at equator | E/W at 23N/S | E/W at 45N/S | E/W at 67N/S
4 	0.0001 	0° 00′ 0.36″ 	individual street, land parcel 	11.132 m 	10.247 m 	7.871 m 	4.3496 m

You are less than centimeter precision. So if you compute with photogrammetry process you could not be less precise than 4m. I'm in doubt with this information writen in exif : 
```
exiftool DSC07492.JPG | grep Accuracy
GPSXY Accuracy                  : 0.037499999161810
GPSZ Accuracy                   : 0.086999997496605
```
Or may be Pix4d use it to correct gps exif value and add precision.

If you want to add precison coordinates have to be 7 digits minimum (0.0000001) or 0° 00′ 0.00036″  Unfortunately, it's how gps coords are saved in exifs and it's not possible that's why adding a file with coordinates is needed.
In Pix4d, you could see, 7 digits is in examples : https://support.pix4d.com/hc/en-us/articles/202558539-Input-Files#labelA1_1&gsc.tab=0

If you want another opensource project where it's debate, here you have with mission planner : https://github.com/ArduPilot/MissionPlanner/issues/1586

uasimagery, 2017-09-13:
Also, is there someway to view the accuracy of the maps created in WebODM? I have attached a screenshot from a processed survey in DroneDeploy, it shows the X,Y, Z and RMSE. Is there a way to see these values in WebODM? Thanks so much for everyones help..

Simon, Im in NZ at the moment but when I get back Ill give you a bell. 3DR LoRa turned up today ;)

[image]

dakotabenjamin, 2017-09-13:
OK I see now that it is a precision issue in the exif. Thank you

KommandorKeen, 2017-09-14:
The big yellow circle is the data put into the EXIF, the smaller orange circle is the position pulled out of the EXIF. The grid is 5m. The yellow circles 1m.
 I use EXIFtool commandline to insert the data at the following precision(Lat, Lon, Ht)
-42.89252914,147.4871013,63.3624
So for me the simplest solution is to insert the position into the exif, set the global estimated position accuracy and ensure that the data is being used. I can do this in photoscan, and pix4D but unsure how to make ODM apply the positioning more rigidly. 

[image]

smathermather, 2017-09-14:
Assuming for a moment that you were able to pass through the more accurate coordinates to ODM, OpenSfM (the underlying library) would have to apply the appropriate weighting factors / constraints to the model initialization and generation, with higher weights on the input camera position. I don't think this is a hard problem, if I understand the OpenSfM approach correctly, this is a parameterization and control script issue, not something deeper.

All that said, I don't have time to dive in yet, but if you do @KommandorKeen, you could bug the folks at the OpenSfM about where to access the weighting factors for camera position modeling / SfM (let them know your use case). Make progress on that, and @dakotabenjamin could help you pass that through to ODM.

KommandorKeen, 2017-09-14:
Thanks @smathermather for the directions in how to progress this.
I recall a post somewher about modifying the opensfm config file.  I  will chase it up.

pierotofy, 2017-09-14:
@uasimagery currently we do not output an estimate of the accuracy; this is something that we should add at some point. I've opened a new issue for that here #286

smathermather, 2017-09-14:
@pierotofy, @uasimagery -- this is on the bucket list for the next few months. Mapillary will be adding reporting features to OpenSfM, which then we'll expose in OpenDroneMap.

johanez, 2018-06-11:
hi @KommandorKeen, I am very interested in using PPK/RTK positions... did you look into the opensfm config?

pierotofy, 2020-04-02:
Closing in favor of https://github.com/OpenDroneMap/ODM/issues/1092

closed 2020-04-02 by pierotofy