WebODM/WebODM #283: Installation Problem

smitch41, 2017-09-09:
Hi
Trying to get WebODM running and all seems well until it gives me an Error stating "No Containers to Start". Perhaps I've missed something in the install..?
[image]

pierotofy, 2017-09-09:
Nop, nothing wrong. Just wait for the downloads to finish and you'll be up and running. 😄

smitch41, 2017-09-13:
Thanks....unfortunately I'm stuck on a vessel in the middle of the ocean...very slow download...will have to wait til I'm back on land.

pierotofy, 2017-09-28:
@smitch41 did you manage to get it working? Can I close this?

smitch41, 2017-10-04:
Yes...all working...thankyou.

closed 2017-10-05 by smathermather