WebODM/WebODM #284: How to direct output from webodm to fileshare

mortentj, 2017-09-13:
Hi
I'm newbie using WebODM with docker on Ubuntu. I wan't to direct output from tasks to fileshare  outside docker container.  Right now my C drive keeps filling up and i don't wan't to free space by deleting tasks. I tried to modify the docker-composer.yml but without success. Any ideas how to solve this

Thanks in advance

pierotofy, 2017-09-13:
Hi, welcome! 😄 

Do the steps outlined in #278 help?

KommandorKeen, 2017-09-14:
@pierotofy helped me get this working see #278 , but this only moves the final output files, not the bits along the way. That said if @uasimagery adopted this storage strategy he would have access to the odm_georeferencing_log.txt which contains an estimate of the final georeference error discussed in #282

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/how-to-direct-output-from-webodm-to-fileshare/40

closed 2017-10-19 by pierotofy