WebODM/WebODM #285: Cross Correlation

faraztehrani, 2017-09-14:
Hi,

Is it possible, in OpenDroneMap, to perform cross correlation on the generated DEMs (reconstructed in different times)? 

If not, do you have a plan to include this in the code?

Thanks,

Faraz

pierotofy, 2017-09-14:
Hi @faraztehrani, welcome!

Currently there's nothing built-in to perform correlation on different products (DEMs, orthophotos, etc.). This would be a cool feature to add, however.

To help contributors understand better the use case, what kind of correlation tools would you be looking to have? A "time" slider that allows visual inspection of the results across time, or something more elaborate?

faraztehrani, 2017-10-07:
Hi Piero,
I know some researchers have used Cosi-Corr to  get the horizontal displacement from areal imagery. Perhaps, that's something to be implemented here.  A displacement vector field between different time intervals would be an ideal tool.

smathermather, 2017-10-07:
Hi @faraztehrani -- something that might interest you is the split-merge branch which is designed to help conflate adjacent datasets in X, Y, and Z to allow for larger projects. The approach depends on finding matches between adjacent datasets and transforming adjacent datasets into a single matching coordinate system.

https://github.com/OpenDroneMap/OpenDroneMap/tree/split-merge

It's a work in progress, but works pretty well as is, and might contain some of the pieces you are interested in.

faraztehrani, 2017-10-07:
Hi Stephen,

Thanks for the quick reaction. I'll take a look at it.

See you soon,

Faraz

smathermather, 2017-10-11:
Great meeting and chatting with you. Cheers!

pierotofy, 2017-10-11:
I'll keep this issue open as an improvement request.

faraztehrani, 2017-10-12:
Steve!  It was great talking with you.

Piero and Steve! Look forward to seeing more from you guys.

closed 2020-04-02 by pierotofy