WebODM/WebODM #288: Issues with orthophoto

viks2015, 2017-09-15:
The generated image is patched properly and has lots for gaps in it, the same is applicable for the generated 3d mesh 
Also how to I import the orthophoto in google earth
[image]

pierotofy, 2017-09-15:
Hi and welcome! 😄 

You could try to increase the `min-num-features` parameter in the task settings (set it to 8000) and see if that makes a difference.

To import the orthophoto in Google Earth you could use `gdal2tiles` http://www.gdal.org/gdal2tiles.html.

We also have plans to automate KML/KMZ export functionality into WebODM, but that feature hasn't been implemented yet. See #126

viks2015, 2017-09-16:
Ok thanks for the help I haved increase the min-num-features parameter and have to wait to see the results. 
I webodm the final orthophoto is placed over google map automatically, now if I take take that photo and try to place on Google Earth Pro it does to get placed at the correct location, how can I do that.
I am able to place orthophoto generated by Agisoft Photoscan over Google earth Pro correctly

KommandorKeen, 2017-09-23:
Looks like the holes are outside the footprint of the drone flight.
You cannot expect to get a map of places the software cannot see and match in multiple images. from multiple angles.

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/issues-with-orthophoto/47

closed 2017-10-19 by pierotofy