WebODM/WebODM #289: Error while processing - Could not start process (spawn /code/run.sh ENOENT)

Marcuzzo77, 2017-09-16:
Goodmorning everyone,
I happened on a pc with windows 10 an error when i start the startup for the processing of the pictures of which I attached screenshots.
Says "Could not start process (spawn /code/run.sh ENOENT)"
[image]

pierotofy, 2017-09-16:
Hi, sorry to hear that you're experiencing issues.

How did you install WebODM (did you use docker, did you use the installer from webodm.org, ... )?

Marcuzzo77, 2017-09-16:
I downloaded the packet from included docker toolkit github etc .. It went all right, it worked, then I wanted to expand virtualbox 5.1 virtual disc as it was not the default 20GB ...
Unfortunately some passes have not been successful, so I thought I would start from scratch reinstalling everything.
Now that's the problem ...

pierotofy, 2017-09-16:
I rolled back some changes, if you run:

`./webodm.sh down && docker pull opendronemap/node-opendronemap && ./webodm.sh start`

Does it work now?

Marcuzzo77, 2017-09-16:
No Piero, it does not work yet

pierotofy, 2017-09-16:
Could you try starting from a fresh install?

Marcuzzo77, 2017-09-16:
I tried to delete everything, the Webodm folder, the virtualbox virtual machine created by the docker, downloading packages from the git, then I followed the procedure

https://github.com/OpenDroneMap/WebODM

Nothing to do ... I continue my tests

Marcuzzo77, 2017-09-16:
Ok I finally solved ....
I followed your advice, uninstalled everything, bought virtual machines, and reinstalled everything. Finally it works.
Thanks Piero for your support

pierotofy, 2017-09-17:
Great! 👍 Thank you for reporting the problem!

closed 2017-10-24 by pierotofy