WebODM/WebODM #291: Problem launching/running

dudman, 2017-09-20:
I'm running archlinux 

and tried 

> git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input
> cd WebODM
> ./webodm.sh start

But get an error trying to start command
`Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting db         ... 
Starting node-odm-1 ... 
Starting webapp     ... error

ERROR: for node-odm-1  Timeout value connect was Timeout(connect=60, read=60, total=None), but it must be an int or float.

ERROR: for db  Timeout value connect was Timeout(connect=60, read=60, total=None), but it must be an int or float.
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 124, in _validate_timeout
    float(value)
TypeError: float() argument must be a string or a number, not 'Timeout'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/docker-compose", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python3.6/site-packages/compose/cli/main.py", line 68, in main
    command()
  File "/usr/lib/python3.6/site-packages/compose/cli/main.py", line 121, in perform_command
    handler(command, command_options)
  File "/usr/lib/python3.6/site-packages/compose/cli/main.py", line 804, in start
    containers = self.project.start(service_names=options['SERVICE'])
  File "/usr/lib/python3.6/site-packages/compose/project.py", line 273, in start
    get_deps,
  File "/usr/lib/python3.6/site-packages/compose/parallel.py", line 70, in parallel_execute
    raise error_to_reraise
  File "/usr/lib/python3.6/site-packages/compose/parallel.py", line 168, in producer
    result = func(obj)
  File "/usr/lib/python3.6/site-packages/compose/project.py", line 257, in start_service
    service_containers = service.start(quiet=True, **options)
  File "/usr/lib/python3.6/site-packages/compose/service.py", line 199, in start
    containers = self.containers(stopped=True)
  File "/usr/lib/python3.6/site-packages/compose/service.py", line 186, in containers
    filters=filters)]))
  File "/usr/lib/python3.6/site-packages/docker/api/container.py", line 191, in containers
    res = self._result(self._get(u, params=params), True)
  File "/usr/lib/python3.6/site-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.6/site-packages/docker/api/client.py", line 189, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "/usr/lib/python3.6/site-packages/requests/sessions.py", line 521, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python3.6/site-packages/requests/sessions.py", line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3.6/site-packages/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3.6/site-packages/requests/adapters.py", line 440, in send
    timeout=timeout
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/connectionpool.py", line 582, in urlopen
    timeout_obj = self._get_timeout(timeout)
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/connectionpool.py", line 309, in _get_timeout
    return Timeout.from_float(timeout)
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 154, in from_float
    return Timeout(read=timeout, connect=timeout)
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 97, in __init__
    self._connect = self._validate_timeout(connect, 'connect')
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 127, in _validate_timeout
    "int or float." % (name, value))
ValueError: Timeout value connect was Timeout(connect=60, read=60, total=None), but it must be an int or float.
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 124, in _validate_timeout
    float(value)
TypeError: float() argument must be a string or a number, not 'Timeout'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/docker-compose", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python3.6/site-packages/compose/cli/main.py", line 68, in main
    command()
  File "/usr/lib/python3.6/site-packages/compose/cli/main.py", line 121, in perform_command
    handler(command, command_options)
  File "/usr/lib/python3.6/site-packages/compose/cli/main.py", line 938, in up
    scale_override=parse_scale_args(options['--scale']),
  File "/usr/lib/python3.6/site-packages/compose/project.py", line 417, in up
    warn_for_swarm_mode(self.client)
  File "/usr/lib/python3.6/site-packages/compose/project.py", line 646, in warn_for_swarm_mode
    info = client.info()
  File "/usr/lib/python3.6/site-packages/docker/api/daemon.py", line 90, in info
    return self._result(self._get(self._url("/info")), True)
  File "/usr/lib/python3.6/site-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.6/site-packages/docker/api/client.py", line 189, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "/usr/lib/python3.6/site-packages/requests/sessions.py", line 521, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python3.6/site-packages/requests/sessions.py", line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3.6/site-packages/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3.6/site-packages/requests/adapters.py", line 440, in send
    timeout=timeout
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/connectionpool.py", line 582, in urlopen
    timeout_obj = self._get_timeout(timeout)
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/connectionpool.py", line 309, in _get_timeout
    return Timeout.from_float(timeout)
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 154, in from_float
    return Timeout(read=timeout, connect=timeout)
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 97, in __init__
    self._connect = self._validate_timeout(connect, 'connect')
  File "/usr/lib/python3.6/site-packages/requests/packages/urllib3/util/timeout.py", line 127, in _validate_timeout
    "int or float." % (name, value))
ValueError: Timeout value connect was Timeout(connect=60, read=60, total=None), but it must be an int or float.
`

pierotofy, 2017-09-20:
Hi, welcome!

This looks like a problem with `requests` (the Python library), which has since released a hotfix. You might need to update the library via:

`sudo pip install -U requests`

See this discussion and related threads for more info: https://github.com/docker/compose/issues/4927

pierotofy, 2017-09-28:
@dudman do you still need help with this?

dudman, 2017-10-01:
thanks for the reminder, I had this as a task to finish and get working... I've got it working on a different computer, also archlinux-antergos... I'll still try to understand the differences between the 2. So will add anything or comment here soon.

jacunaf, 2017-10-11:
Hello

This message appears after starting (StartWebODM) and stays this way for hours:

"latest: Pulling from opendronemap/node-opendronemap"

pierotofy, 2017-10-11:
@jacunaf how fast is your internet connection? The program needs to download ~4GB of data during that step.

jacunaf, 2017-10-11:
uf.. very poor
voy a conseguir una conexión mejor
gracias

Arq. Jorge Acuña Fretes
+595 991 855900

closed 2017-10-19 by pierotofy