WebODM/WebODM #294: Failed to execute script docker-compose

MyNameIsAina, 2017-09-29:
I have Windows 10 Home, virtualization is activated
I have installed Docker Toolbox, and I'm trying to execute a very simple project 
The error occures Both in Docker Quickstarter and simple cmd while I'm running the command: docker-compose up
Plz help
the error:

Traceback (most recent call last):
  File "docker-compose", line 3, in <module>
  File "compose\cli\main.py", line 68, in main
  File "compose\cli\main.py", line 118, in perform_command
  File "compose\cli\main.py", line 928, in up
  File "compose\project.py", line 414, in up
  File "compose\project.py", line 640, in warn_for_swarm_mode
  File "site-packages\docker\api\daemon.py", line 90, in info
  File "site-packages\docker\utils\decorators.py", line 46, in inner
  File "site-packages\docker\api\client.py", line 189, in _get
  File "site-packages\requests\sessions.py", line 488, in get
  File "site-packages\requests\sessions.py", line 475, in request
  File "site-packages\requests\sessions.py", line 596, in send
  File "site-packages\requests\adapters.py", line 423, in send
  File "site-packages\requests\packages\urllib3\connectionpool.py", line 595, in urlopen
  File "site-packages\requests\packages\urllib3\connectionpool.py", line 363, in _make_request
  File "httplib.py", line 1042, in request
  File "httplib.py", line 1082, in _send_request
  File "httplib.py", line 1038, in endheaders
  File "httplib.py", line 882, in _send_output
  File "httplib.py", line 844, in send
  File "site-packages\docker\transport\npipeconn.py", line 31, in connect
  File "site-packages\docker\transport\npipesocket.py", line 22, in wrapped
  File "site-packages\docker\transport\npipesocket.py", line 50, in connect
pywintypes.error: (2, 'WaitNamedPipe', 'The system cannot find the file specified.')
Failed to execute script docker-compose

pierotofy, 2017-09-29:
Hi, does #137 help?

Also, from the virtualbox manager application, can you confirm that the "default" virtual machine is running?

MyNameIsAina, 2017-09-29:
@pierotofy  yea the virtual machine is running

pierotofy, 2017-09-29:
Mm, I'm not sure what the cause of the error would be, but it's related to docker. I've seen this problem come up a few times already, I would start searching the docker forums to see if there's a solution (maybe upgrade docker-compose to the latest version, or configure something)?

pierotofy, 2017-09-29:
https://www.google.com/search?q=WaitNamedPipe+windows+10+home+docker-compose&oq=WaitNamedPipe+windows+10+home+docker-compose&aqs=chrome..69i57j0.3843j0j1&sourceid=chrome&ie=UTF-8

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/failed-to-execute-script-docker-compose/38

closed 2017-10-19 by pierotofy