WebODM/WebODM #296: Test individual React.js components in more depth

pierotofy, 2017-10-02:
Currently the frontend unit tests only do a basic smoke test. 

Functionality should be tested more in depth, especially for the critical areas such as dashboard and upload/task management.

alejandronanez-huge, 2017-10-02:
Hey @pierotofy, do you have any _goal_ for what needs to be done in terms of tests? It can be quite challenging to achieve 100% of code coverage.

pierotofy, 2017-10-02:
Yes absolutely! 100% code coverage is probably not what we want at the moment, only more coverage for some of the critical sections.

Most important components are probably:
 - Dashboard (https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/tests/Dashboard.test.jsx)
 - TaskList (https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/components/tests/TaskList.test.jsx)
- EditTaskForm (https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/components/tests/EditTaskForm.test.jsx)

pierotofy, 2017-10-02:
So testing the basic "happy path" of a user going through the process of uploading images, checking the processing status and being able to navigate to map view / 3d view once processing is completed.

pierotofy, 2021-02-03:
This is likely never going to happen.

closed 2021-02-03 by pierotofy