WebODM/WebODM #297: Multispectral geotiff images images compatibility (Micasense RedEdge)

agustinbus, 2017-10-04:
I have a use a Micasense RedEdge that take five images every shoot, one for each band (R G B NIR RedEdge). Each len in When i use multispectral tiff images as input for processing i get the following error:

> [ERROR]   Not enough supported images in /var/www/data/

and this message:

> An orthophoto could not be generated. To generate one, make sure GPS information is embedded in the EXIF tags of your images, or use a Ground Control Points (GCP) file. 

There is any way to run WebODM with multispectral images (each band in a separate image)?

Thanks in advance, any help is appreciated.

juvinski, 2017-10-04:
Hi @agustinbus,

could you post some pictures to test here ?
if you run gdalinfo <picture> what results you have ?

best regards

dakotabenjamin, 2017-10-04:
OpenDroneMap currently does not support multispectral tiff images. It's on our list, though.

agustinbus, 2017-10-04:
Hi @juvinski @dakotabenjamin , thank for your fast response!

@juvinski I do not have in my pc right now, but later i will post a dropbox link with a Micasense RedEdge pictures to serve as an example. I will post gdalinfo output too.

@dakotabenjamin :( I do not have much knowledge in python but i will investigate that. I think converting from tiff to jpeg will not be a big deal, but perform the simultaneous processing of each band (image) for accurate calculations like NDVI will be a problem.

dakotabenjamin, 2017-10-04:
That would be great! There's some threads already over at [OpenDroneMap](https://github.com/opendronemap/opendronemap) just search tiff or multispectral.

Some of the issues I think will come up is alignment, since each band is slightly different.

juvinski, 2017-10-04:
And for every camera the order of the channels inside the pictures can be different.
@dakotabenjamin , I'm working with mosaic alignment how we discuss on gitter, when i had something more accurate I will call you again.
This model from Micasense can be important to do a more modular development - today will just work for my scenario.

egenial, 2018-09-07:
News for this feature? 😍 i have one micanse redEdge-m. Like process on open Drone map.

smathermather, 2018-09-25:
It's something being discussed, but no one yet has time or money to implement. Follow this thread for more in depth info on what is required: https://github.com/OpenDroneMap/OpenDroneMap/issues/500

smathermather, 2018-09-25:
(Also, there are some functional workarounds posted there until ODM directly supports this).

MatthiasSiewert, 2019-05-13:
Support for Parrot Sequoia cameras would also be great. It is one of the more commonly used multispectral sensors.

pierotofy, 2019-12-03:
I know this is an old discussion, however, efforts are underway to implement multispectral support! :clinking_glasses: @agustinbus do you have some example images that we could use for testing?

samsalway, 2019-12-06:
Hi @pierotofy, I'm able to provide images from Parrot Sequoia, would that help? I'd love to start using ODM if it supported multispectral and Sequoia. I'm a software engineer (10+ years) and would be happy to help where I can.

jonnyforestGIS, 2019-12-06:
@pierotofy I can try to speak with my colleagues and Institute if we can share an example dataset from a micasense Altum. Can be used as a test?
Kind regards,
João Gaspar

pierotofy, 2019-12-06:
Hey @samsalway :hand: yes the dataset would help. We would also love code contributions on any of these issues: https://github.com/OpenDroneMap/WebODM/issues?q=is%3Aopen+is%3Aissue+label%3A%22help+wanted%22

@jonnyforestGIS yes would be helpful!

samsalway, 2019-12-07:
[Sequoia sample data](https://cloud.salway.net/s/zWy3T23A8beoCtP) 63MB, this link will expire in 2 weeks.

jonnyforestGIS, 2019-12-11:
@pierotofy 
Dataset sample of a flight with a Micasense Altum (with all bands. Notice: The termal band have a different resolution, I don't know if you want to test this as well).

These sample images were provided from my colleague André Duarte @DuarteAndre. He is working on the field with the UAV. See: [Forum WebODM](https://community.opendronemap.org/t/request-for-datasets-shot-with-sequoia-multispec-4c-and-micasense-rededge/2922/2)

Flight height: 120 meters
Foward Overlapping: 70%
Lateral Overlapping: 70%
UAV Speed: 7m/s

[Sample 600MB](https://portucelsoporcel-my.sharepoint.com/personal/andre_duarte_thenavigatorcompany_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fandre%5Fduarte%5Fthenavigatorcompany%5Fcom%2FDocuments%2FSET%5FMICASENSE&originalPath=aHR0cHM6Ly9wb3J0dWNlbHNvcG9yY2VsLW15LnNoYXJlcG9pbnQuY29tLzpmOi9nL3BlcnNvbmFsL2FuZHJlX2R1YXJ0ZV90aGVuYXZpZ2F0b3Jjb21wYW55X2NvbS9FbHBDUVA3aWhPSkRxblRBNjJuVFhjb0J6UXRlanMyTFhpc0t2TmVOYndfMWtBP3J0aW1lPUJSMWZFQjUtMTBn)

pierotofy, 2019-12-11:
Thanks @jonnyforestGIS, but for some reason I cannot open the link? Says my user is not in the organization directory.

jonnyforestGIS, 2019-12-11:
@pierotofy hum ok I think @DuarteAndre forget to open to external. I will speak with them and update this.

jonnyforestGIS, 2019-12-11:
@pierotofy sorry this can be some kind of permission issue on the sharing. Perhaps Andre don't put share with external. Let me ask André because I just redirect the link.

jonnyforestGIS, 2019-12-12:
@pierotofy can you check again the link in the comment above?

pierotofy, 2019-12-12:
[image]

jonnyforestGIS, 2019-12-12:
@pierotofy sorry this is not easy. :sweat_smile: Please check again. I think the problem is permission on the folder but not in the zip file.

jonnyforestGIS, 2020-01-21:
Hi @pierotofy. Sorry for the late reply but did you get the correctly access to the shared folder and download the dataset in the last time? 
Regards

pierotofy, 2020-01-21:
Hey @jonnyforestGIS, unfortunately not; still same error.

[image]

jonnyforestGIS, 2020-01-22:
@pierotofy try this. I recreate the shareable link. Sorry for the trouble.
[Donwload Link](https://portucelsoporcel-my.sharepoint.com/:u:/g/personal/joao_filipe_gaspar_thenavigatorcompany_com/EQpw0oAp2L9Bj8zMyWPX-NcBkU_IiTNz8cLk9psGAw6gog?e=Bm2eIU)

pierotofy, 2020-01-22:
Interesting; is it normal that the LWIR images ( for example IMG_0024_6.tif ) are only 160x120?

jonnyforestGIS, 2020-01-22:
Hi @pierotofy yes. Is the termal band. Micasense Altum collect with a different resolution compared to the others bands.

jonnyforestGIS, 2020-02-20:
@pierotofy this is already in the latest webODM?
https://www.opendronemap.org/2020/02/odm-0-9-8-adds-multispectral-16bit-tiffs-support-and-moar/

Do you need some test and feedback?

pierotofy, 2020-02-20:
Yep, this is already in WebODM. (And this issue should be closed).

Testing would be great; post feedback on the forum?

closed 2020-02-20 by pierotofy