WebODM/WebODM #301: Process exited with code 1

WildcatOne, 2017-10-13:
Trying to process 138 images, and after 4 hours, it fails.
Any ideas? here is the message:
Created on: 10/13/2017, 1:43:54 PM
Status: Failed
Options: mesh-octree-depth: 12, min-num-features: 8000, orthophoto-resolution: 60, dem-resolution: 0.04, use-exif: true, skip-resize: true, dsm: true, use-25dmesh: true

2017-10-13 19:54:13,486 Merging depthmaps

Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 128, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

WildcatOne, 2017-10-13:
Here are the images if anyone wants them.
https://www.dropbox.com/sh/b9avcja8itzml6o/AADebNZWA13rMMuH-In2Be0Ta?dl=0

smathermather, 2017-10-14:
Do they have gps in the exif tags?

WildcatOne, 2017-10-16:
Yes, they do.
[image]

pierotofy, 2017-10-16:
Hi, this might be a problem related to memory usage. What happens if you reduce `min-num-features` to 4000 and remove `skip-resize`? Also, could you share some information about your system (amount of RAM, CPU, operating system, are you running with docker, how much memory have you allocated to the docker VM, etc.)?

closed 2017-10-16 by WildcatOne