WebODM/WebODM #302: Process exited with code 1 using odm_data_benchmark dataset

Jorge-Mendes, 2017-10-15:
Hello, this is my first experience with OpenDroneMap/WebODM and it did not go well. I was trying to process the "odm_data_benchmark" dataset (https://github.com/OpenDroneMap/odm_data_benchmark) but I got the following error:

> "Process exited with code 1" means that part of the processing failed. Try tweaking the Task Options as follows:
> 
>    Increase the min-num-features option, especially if your images have lots of vegetation
>    Enable the use-pmvs option.

I have already increased the "min-num-features" option from 4000 (default) to 10000 and enabled "use-pmvs", but the problem persists.

I noticed that the images were duplicated (the same image with the .jpg and .JPG extension). I deleted the ones with the extension .JPG and I only had the .jpg extension, but the problem continues.

The task output was as follows:
[task_output_odm_data_benchmark.txt](https://github.com/OpenDroneMap/WebODM/files/1385778/task_output_odm_data_benchmark.txt)

I tested the "odm_data_aukerman" dataset (https://github.com/OpenDroneMap/odm_data_aukerman) and there was no problem.

Any idea what the reason and possible solution?

dakotabenjamin, 2017-10-16:
The reason is that it is not a good dataset, there's no way to reconstruct it. 

I know it's misleading to call it benchmark, I thought I had removed that dataset from the repo.

pierotofy, 2017-10-19:
Seems like we can close this then.

closed 2017-10-19 by pierotofy