WebODM/WebODM #303: Indexing in utm_pole index

KassieC, 2017-10-16:
I closed the previous issue #267 since the recent fixes seemed to work. However, I've found webODM is exiting at a different step with a similar-but-not-identical error.

> Traceback (most recent call last):
>   File "/code/run.py", line 47, in <module>
>     plasm.execute(niter=1)
>   File "/code/scripts/odm_georeferencing.py", line 164, in process
>     geo_ref.parse_coordinate_system(tree.odm_georeferencing_coords)
>   File "/code/opendm/types.py", line 317, in parse_coordinate_system
>     self.utm_pole = ref[2][len(ref) - 1]
> IndexError: string index out of range

I thought the change over to the use of exiv2 in pull request [#675](https://github.com/OpenDroneMap/OpenDroneMap/pull/675). Might have fixed this but after `./webodm.sh update` it kicks me out with the above error. I have a couple of photosets up over in #267 with Alaskan geotags.

pierotofy, 2017-10-16:
Thanks for reporting the issue; could you link the exact dataset you are using to reproduce this problem (I know it's in #267, just want to make sure we have the right one)?

KassieC, 2017-10-16:
Hi Piero, [I've uploaded a smaller chunk of another data set](https://drive.google.com/open?id=0B3f3_q9Y7VunXzRuTW1kMTE5MDA) that should get to the error faster. However, for the more patient people, my [Little Granite Creek](https://drive.google.com/open?id=0B3f3_q9Y7VuncnhfTmdTdUVHeTg) from issue #267 or [Archangel Valley](https://drive.google.com/open?id=0B3f3_q9Y7VunYXdlZDdoYXhFZnM) sets also fail out in an identical fashion.

Wes-Skeeter, 2017-10-20:
I have run into the exact same issue processing images in UTM 8 at ~69 degrees north in the NWT.  Here is a small subset of [my dataset](https://drive.google.com/file/d/0B7GsdLnhj_QfM2pucG5PbVNGTXc/view?usp=sharing)

Running:

`docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap --mesh-size 1000000 --force-ccd 8.8`

throws this error:

`[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /code/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -inputCoordFile /code/odm_georeferencing/coords.txt -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt
[DEBUG]   Line: WGS84 UTM 8N
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 164, in process
    geo_ref.parse_coordinate_system(tree.odm_georeferencing_coords)
  File "/code/opendm/types.py", line 317, in parse_coordinate_system
    self.utm_pole = ref[2][len(ref) - 1]
IndexError: string index out of range
`

Wes-Skeeter, 2017-10-27:
So, I'm actually using OpenDroneMap, not web odm, but maybe this will be helpful anyway.  I ran into a similar problem, using UTM zone 8N.

Tweaking two lines in /code/opendm/types.py and fixed the issue for me:
I changed
 ```
           line = f.readline()
            log.ODM_DEBUG('Line: %s' % line)
            ref = line.split(' ')
            # match_wgs_utm = re.search('WGS84 UTM (\d{1,2})(N|S)', line,https://www.facebook.com/ re.I)
            if ref[0] == 'WGS84' and ref[1] == 'UTM':  # match_wgs_utm:
                self.datum = ref[0]
                self.utm_pole = str(ref[2][-2:-1]) #   --- used to be -  ref[2][len(ref) - 1]
                self.utm_zone = int(ref[2][:-2]) #    --- used to be - ref[2][:len(ref[2]) - 1])
```

for some weird reason, working with UTM 8N, 
len(ref[2]) = 3 when it should be  2 ie. ref[2] = '8N ', not '8N'.  I played around w/ the aukerman test dataset which is in UTM 17N and had the same issue ref[2] = '17N ', not '17N'

If that doesn't work, another possible "fix" would be to just manually set:`
`self.utm_pole = 'N'`
since you know beforehand which coordinate system you should be working in

pierotofy, 2017-10-27:
@Wes-Skeeter are you running the latest copy of ODM? I noticed that you had:

`line = f.readline()`

Whereas it should be:

https://github.com/OpenDroneMap/OpenDroneMap/blob/master/opendm/types.py#L311

(The rstrip call should have fixed this problem)

Wes-Skeeter, 2017-10-27:
@pierotofy  After I first encountered the problem while trying to run ODM through docker, I downloaded [v 0.3.1](https://github.com/OpenDroneMap/OpenDroneMap/releases) to have a look myself and try to fix it.  I downloaded it on Wednesday I think?  Got to the link from [here](http://opendronemap.org/).  rstirp was not in that code.  The little fix above has solved the problem for me but rstrip works too.  Thanks!

closed 2017-11-20 by KassieC