WebODM/WebODM #304: [WARNING] Could not find ccd_width in file. Use --force-ccd

sellinios, 2017-10-16:
Hi, 

I have a phantom Pro 4 drone and I am getting the following message, 

[WARNING] Could not find ccd_width in file. Use --force-ccd 

How do I set up the ccd manually ? 

Could you please help me ?

pierotofy, 2017-10-18:
Hi, if you know the value for your camera sensor, you can set it by modifying the default preset before submitting a task for processing:

[image]

[image]

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/warning-could-not-find-ccd-width-in-file-use-force-ccd/39

closed 2017-10-19 by pierotofy