WebODM/WebODM #305: What would be the best way to run WebODM entirely on an iPad?

FarhadG, 2017-10-16:
I'd like to run WebODM entirely on an iPad as a standalone local unit -- i.e. no web access. In doing so, what are some techniques to help an iPad be able to generate the 3D models and render as efficiently as possible? Offload to external local processors (if possible or feasible)?

pierotofy, 2017-10-17:
Hey @FarhadG, while it would be really cool, it's not possible to run the entire ODM stack solely on an iPad. With that said, you could install WebODM on a small portable server (see for example how the team at POSM did it: http://theoutpost.io/posm/hardware.html) and use the web interface from your iPad.

FarhadG, 2017-10-17:
Thanks for the quick response, @pierotofy. I assume it's due to RAM being an issue on the iPad?

Given we have little to no connectivity onsite, we're trying to figure out ways for generating a low-res (preview mode) model onsite in the least amount of time.

Curious if others have gone down this path?

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/what-would-be-the-best-way-to-run-webodm-entirely-on-an-ipad/43

closed 2017-10-19 by pierotofy