WebODM/WebODM #306: Generation Surface Model failed

Marcuzzo77, 2017-10-18:
I have a problem with the generation of the surface model. After processing I can not generate it. I attach pictures
[image]
[image]

pierotofy, 2017-10-18:
Hi, did you pass the "--dsm" option when processing the task (should be in the "Default" preset)?
Also, could you copy/paste the output of your processing task? How did you install WebODM (docker, native install, etc.)?

Marcuzzo77, 2017-10-18:
Hi Piero,

actually use WebODM on different operating systems. On Windows 10 Pro I use the docker with the virtual Hyper V machine by default and I have no problem with Ubuntu 17.04 instead ...
When I go to select DSM + DTM, the option you choose is not preserved when you save the options, but always use the "Default" option automatically.
Some parameters such as dsm check and dtm check are missing from the options list on Ubuntu 17.04 ...

Marcuzzo77, 2017-10-18:
Hi Piero,

I fixed the problem with an update (./webodm.sh update) at the latest available version. Thank you for your usual collaboration

closed 2017-10-18 by pierotofy