WebODM/WebODM #307: Processing error when I try over 151 pictures, the first 101 work fine

LRDquarry, 2017-10-19:
I have tried to increase virtual box memory to just under 6 meg (and 2 for my windows 10 home host).
I have read and also tried various options to process the full set (218 I think);
Force_CCD (is 8.8 correct for a Phantom 4 Pro), Min_Num_Features 10000, and use PMVS because it is heavy forest.

Here is the process info:
[Uploading task_output.txt…]()

and the first few pictures
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

pierotofy, 2017-10-19:
Hi, unfortunately 6GB is not a whole lot of memory for the number of images you are trying to process, especially with PMVS. I would try to modify your VirtualBox settings to decrease the number of CPUs to one. This will slow down processing, but should reduce memory usage as well.

LRDquarry, 2017-10-19:
On first look, it appears to be set to 1.
Btw, should the Virtualbox interface remain open after i shut down WebODM? I always need to restart the computer to make it clear out.

pierotofy, 2017-10-19:
You can safely close the Virtualbox interface.

If you remove the pmvs flag, are you able to process the images? Is it just that the trees look "bulky" with the default?

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/processing-error-when-i-try-over-151-pictures-the-first-101-work-fine/41

closed 2017-10-19 by pierotofy

LRDquarry, 2017-10-20:
I was trying only with defaults for 101 and 150. Have since run on 125 but
it fails too.

Ultimately I am trying to use drone pics to identify and locate trees for
forestry application. I tried dronedeploy and precision hawk, decided I
need more control and access.

Is there a better way that you (or anyone else) know of?
I am even prepared to pay for tutoring or coding the workflow.