WebODM/WebODM #310: Integrate --rerun parameter

tobi-kellner, 2017-10-30:
[This is more of a feature request, apologies if this is the wrong place for them, but there seem toi be a few]
In "naked" ODM I can re-run individual processing tasks using the **--rerun parameter**.
Using WebODM, I cannot seem to do that. 
This is unfortunate in my scenario as I am battling with crashes, mostlikely due to insufficient RAM, in the OdmMeshing process - around 7h into the process. I would like to experiment with different settings to see how they influence behaviour during this task but it is a pain if I have to run everything from the beginning every time.
Similarly, experimenting with different DEM/DSM parameters is easier if you don't have to run everything from step 0.

In the longer run, what would be awesome would be a bit more info on the order and dependencies of taks/steps, including which parameters influence what.
It's probably obvious to you all, but for noobs like me it is not always clear which tasks build upon each other, and which parameters have effects "downstream".
Provided the raw information, I would happily help writing this up or turning it into a table or diagram.

Thank you for your amazing work!

Greetings from Freiburg, Germany,
Tobi

pierotofy, 2017-10-30:
Hey Tobi, thanks for the feedback. This feature has been on the radar for a while, see #229. I think this will probably be available sometimes early next year.

tobi-kellner, 2017-10-30:
@pierotofy, thank you! 
And apologies for the double-post. Reminds me to always search first...

Btw, from a newbie perspective, it is a bit confusing what is currently the best place for raising issues and asking questions. 
There is this forum, then there is
http://community.opendronemap.org/c/webodm
and then there is the chat at
https://gitter.im/OpenDroneMap/generalhelp

I want to be good and play by the rules, and not ask stupid questions in the wrong places, but it's not so easy to figure out what these are :)

Thanks a lot for your help,

Tobi

pierotofy, 2017-10-30:
No worries, we're in the process of phasing out the gitter chat and slowly pushing the forum at community.opendronemap.org.

Issues (bugs, faults) go here (Github)
Everything else goes on community.opendronemap.org/c/webodm

pierotofy, 2018-01-26:
This has now been implemented as part of #380. Thanks again for the feature request!

closed 2018-01-26 by pierotofy