WebODM/WebODM #313: Can't start WebODM

Yes21, 2017-11-04:
I've installed WebODM on Ubuntu 17.04 with Docker.

When I try to start it (./webodm.sh start), I have the following output : 

Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting db         ... 
Starting node-odm-1 ... 
Starting webapp     ... error

ERROR: for db  ('Connection aborted.', error(13, 'Permission denied'))

ERROR: for node-odm-1  ('Connection aborted.', error(13, 'Permission denied'))
ERROR: Couldn't connect to Docker daemon at http+docker://localunixsocket - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.
ERROR: Couldn't connect to Docker daemon at http+docker://localunixsocket - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.

Could you help me to solve this problem ?

YveS

pierotofy, 2017-11-05:
Hi @Yes21, this is probably a problem related to how you setup docker. Is your user part of the `docker` group? Also, make sure you log-in and out after you've added the group to your user.

pierotofy, 2017-11-05:
`sudo adduser <youruser> docker`

Yes21, 2017-11-05:
That was the problem. Now all is ok, and after a lot of downloads, I can now start WebODM in my browser. I will test it with my own georeferenced pictures to create orthomosaic.
Many thanks for tour help.
YveS

closed 2017-11-06 by pierotofy