WebODM/WebODM #314: Help Please:  /usr/local/lib/python2.7/dist-packages/loky/backend/semaphore_tracker.py:170: UserWarning: semaphore_tracker: There appear to be 1 leaked semaphores to clean up at shutdown   len(cache))

Enabran, 2017-11-05:
Thanks in advance for any help.

Install: I installed WebODM via docker on Ubuntu, i7 w/ 14GB RAM
Browser: Firefox 54 on Ubuntu Linux
Problem: Processing 439 images produces the following error:

/usr/local/lib/python2.7/dist-packages/loky/backend/semaphore_tracker.py:170: UserWarning: semaphore_tracker: There appear to be 1 leaked semaphores to clean up at shutdown
  len(cache))
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 115, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 135

Is this a memory leak or do i lack sufficient memory for the task? 
[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/1443748/task_output.txt)
[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/1443750/task_output.txt)

pierotofy, 2017-11-05:
I'm going to guess that you lack sufficient memory for the task. Does processing work if you process a smaller subset of the images (say, 100 images)?

Enabran, 2017-11-14:
Thanks for the input. I have upgraded the RAM in my machine to 32GB.  I still get this error about 2 out of every 8  processing jobs.   The error dose not seem to be dependent on the number of images processed.

pierotofy, 2017-12-22:
Hey @Enabran, have you figured out if the error is consistently reproducible or if it's tied to a particular variable?

PeterSprague, 2018-01-29:
I'm also getting a very similar error  Running as ODM node.

Docker images work fine on an amd64 workstation w/ 32 cores & 64Gb ram.  

Tuning & testing for lower resource hardware
On packet.net type-0 4 core atom w/ 8Gb ram
docker --limit-mem 7G --> so to leave rm for OS
157 images
see: https://github.com/OpenDroneMap/WebODM/issues/314
settings.yaml - opensfm @ 2
@01:56 of run
2018-01-28 23:52:13,033 Robust matching time : 0.000956058502197s
2018-01-28 23:52:13,033 Full matching 264 / 279, time: 9.20634508133s
2018-01-28 23:52:22,074 2017_0909_115521_230.JPG - 2017_0909_115558_248.JPG has 397 candidate matches
2018-01-28 23:52:22,076 Robust matching time : 0.00139093399048s
2018-01-28 23:52:22,076 Full matching 360 / 397, time: 9.04226994514s
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm create_tracks /var/www/data/0f1e05af-76f5-4cf5-984b-e5a1fef84764/opensfm
2018-01-28 23:52:23,363 reading features
2018-01-28 23:52:43,032 Merging features onto tracks
2018-01-28 23:53:14,937 Good tracks: 169240
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /var/www/data/0f1e05af-76f5-4cf5-984b-e5a1fef84764/opensfm
2018-01-28 23:54:05,452 Starting incremental reconstruction
2018-01-28 23:55:25,230 Starting reconstruction with 2017_0909_114819_028.JPG and 2017_0909_114823_030.JPG
2018-01-28 23:55:25,231 Common tracks: 5438
/usr/local/lib/python2.7/dist-packages/loky/backend/semaphore_tracker.py:170: UserWarning: semaphore_tracker: There appear to be 1 leaked semaphores to clean up at shutdown
  len(cache))
Illegal instruction (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 115, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132

settings.yaml - opensfm @ 1
@03:36 of run
2018-01-29 06:52:00,566 Robust matching time : 0.00222086906433s
2018-01-29 06:52:00,566 Full matching 113 / 133, time: 8.93961906433s
2018-01-29 06:52:09,607 2017_0909_115521_230.JPG - 2017_0909_115532_236.JPG has 282 candidate matches
2018-01-29 06:52:09,608 Robust matching time : 0.000813007354736s
2018-01-29 06:52:09,609 Full matching 272 / 282, time: 9.04200792313s
2018-01-29 06:52:18,582 2017_0909_115521_230.JPG - 2017_0909_115558_248.JPG has 404 candidate matches
2018-01-29 06:52:18,584 Robust matching time : 0.00192189216614s
2018-01-29 06:52:18,584 Full matching 365 / 404, time: 8.9751970768s
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm create_tracks /var/www/data/e3586734-55ee-41b6-adce-a6a9ad815891/opensfm
2018-01-29 06:52:19,683 reading features
2018-01-29 06:52:39,300 Merging features onto tracks
2018-01-29 06:53:11,229 Good tracks: 169237
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /var/www/data/e3586734-55ee-41b6-adce-a6a9ad815891/opensfm
2018-01-29 06:54:02,753 Starting incremental reconstruction
2018-01-29 06:55:33,190 Starting reconstruction with 2017_0909_114819_028.JPG and 2017_0909_114823_030.JPG
2018-01-29 06:55:33,191 Common tracks: 5239
Illegal instruction (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 115, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132

Enabran, 2018-01-30:
The issue went away when I moved the application to another machine. New machine has dual xeon processors , 40gb of ram, and a solid state drive.  this setup works very well.

closed 2018-01-30 by pierotofy