WebODM/WebODM #315: Problem running Webodm

DroneMapper, 2017-11-07:
I installed WebODM with Docker Toolbox and Python 3.6.
I am using firefox as browser but firefox give me a error as below without giving me the option to log in

Environment:

Request Method: GET
Request URL: http://192.168.99.100:8000/

Django Version: 1.11.1
Python Version: 3.5.2
Installed Applications:
['django.contrib.admin',
 'django.contrib.auth',
 'django.contrib.contenttypes',
 'django.contrib.sessions',
 'django.contrib.messages',
 'django.contrib.staticfiles',
 'django.contrib.gis',
 'django_filters',
 'guardian',
 'rest_framework',
 'rest_framework_nested',
 'webpack_loader',
 'corsheaders',
 'app',
 'nodeodm']
Installed Middleware:
['corsheaders.middleware.CorsMiddleware',
 'django.middleware.security.SecurityMiddleware',
 'django.contrib.sessions.middleware.SessionMiddleware',
 'django.middleware.common.CommonMiddleware',
 'django.middleware.csrf.CsrfViewMiddleware',
 'django.contrib.auth.middleware.AuthenticationMiddleware',
 'django.contrib.messages.middleware.MessageMiddleware',
 'django.middleware.clickjacking.XFrameOptionsMiddleware',
 'django.middleware.locale.LocaleMiddleware']

Traceback:

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  65.                 return self.cursor.execute(sql, params)

The above exception (relation "auth_user" does not exist
LINE 1: SELECT COUNT(*) AS "__count" FROM "auth_user" WHERE "auth_us...
                                          ^
) was the direct cause of the following exception:

File "/usr/local/lib/python3.5/site-packages/django/core/handlers/exception.py" in inner
  41.             response = get_response(request)

File "/usr/local/lib/python3.5/site-packages/django/core/handlers/base.py" in _get_response
  187.                 response = self.process_exception_by_middleware(e, request)

File "/usr/local/lib/python3.5/site-packages/django/core/handlers/base.py" in _get_response
  185.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/webodm/app/views.py" in index
  20.     if User.objects.filter(is_superuser=True).count() == 0:

File "/usr/local/lib/python3.5/site-packages/django/db/models/query.py" in count
  364.         return self.query.get_count(using=self.db)

File "/usr/local/lib/python3.5/site-packages/django/db/models/sql/query.py" in get_count
  499.         number = obj.get_aggregation(using, ['__count'])['__count']

File "/usr/local/lib/python3.5/site-packages/django/db/models/sql/query.py" in get_aggregation
  480.         result = compiler.execute_sql(SINGLE)

File "/usr/local/lib/python3.5/site-packages/django/db/models/sql/compiler.py" in execute_sql
  886.             raise original_exception

File "/usr/local/lib/python3.5/site-packages/django/db/models/sql/compiler.py" in execute_sql
  876.             cursor.execute(sql, params)

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  80.             return super(CursorDebugWrapper, self).execute(sql, params)

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  65.                 return self.cursor.execute(sql, params)

File "/usr/local/lib/python3.5/site-packages/django/db/utils.py" in __exit__
  94.                 six.reraise(dj_exc_type, dj_exc_value, traceback)

File "/usr/local/lib/python3.5/site-packages/django/utils/six.py" in reraise
  685.             raise value.with_traceback(tb)

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  65.                 return self.cursor.execute(sql, params)

Exception Type: ProgrammingError at /
Exception Value: relation "auth_user" does not exist
LINE 1: SELECT COUNT(*) AS "__count" FROM "auth_user" WHERE "auth_us...

pierotofy, 2017-11-07:
Hey @DroneMapper, looks like for some reason your database did not start, or wasn't able to connect to the rest of the application.

Could you output a complete log of:

`./webodm.sh start`

?

DroneMapper, 2017-11-07:
ownloads/              Pictures/
Favorites/              PrintHood@
IntelGraphicsProfiles/  Recent@
Links/                  'Saved Games'/
'Local Settings'@       Searches/
Music/                  SendTo@
'My Documents'@         'Start Menu'@
NetHood@                Templates@
NTUSER.DAT              Videos/
ntuser.dat.LOG1         WebODM/

drone@LAPTOP-19S0Q02U MINGW64 ~
$ cd webodm

drone@LAPTOP-19S0Q02U MINGW64 ~/webodm (master)
$ ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting db         ... done
Starting node-odm-1 ... done
Starting webapp     ... done

drone@LAPTOP-19S0Q02U MINGW64 ~/webodm (master)
$

DroneMapper, 2017-11-07:
Message in Firefox

Request Method: | GET
-- | --
http://192.168.99.100:8000/
1.11.1
OperationalError
FATAL:  the database system is starting up
/usr/local/lib/python3.5/site-packages/psycopg2/__init__.py in connect, line 164
/usr/local/bin/python3.5
3.5.2
['/webodm',  '/usr/local/bin',  '/webodm',  '/usr/local/lib/python35.zip',  '/usr/local/lib/python3.5',  '/usr/local/lib/python3.5/plat-linux',  '/usr/local/lib/python3.5/lib-dynload',  '/usr/local/lib/python3.5/site-packages']
Tue, 7 Nov 2017 17:36:11 +0000

pierotofy, 2017-11-07:
Try to run:

```
docker exec -ti $(docker ps -q --filter "name=webapp") bash -c "cd /webodm; python manage.py makemigrations; python manage.py migrate"
```

DroneMapper, 2017-11-07:
Thanks for the help, it works

closed 2017-11-07 by pierotofy

DroneMapper, 2017-11-07:
It works but wants me to add a processing node

pierotofy, 2017-11-07:
Add the following from the "Add processing node" panel:

Hostname: node-odm-1
Port: 3000

BruceParmele, 2017-11-16:
@pierotofy  Your solution worked for me.  Thanks.