WebODM/WebODM #316: WebODM AR Mobile App

pierotofy, 2017-11-07:
Create a iOS or Android (or both) app able to:
 - Log-in to a WebODM instance
 - Pull list of processed tasks
 - Allow the user to select a textured 3D model
 - Select a flat area onto which to project the model
 - Allow the user to navigate the model in AR

 - Bonus: allow measurements, display point clouds as well.

gl3nda85, 2017-12-19:
Hey i would be interesting in working on this in my spare time, have you begun??

pierotofy, 2017-12-19:
Barely, I've started a react-native app skeleton at https://github.com/pierotofy/webodm-ar. I'm currently doing work on the Expo framework to add support for plane detection (it's currently missing from Expo THREE). Once that's added, the app can be built. In the meanwhile there are non-AR things that could be worked on, such as login screen, project list selection, etc.

Would love to collaborate if you're interested!

gl3nda85, 2017-12-30:
hey dude, sorry for the late reply, i actually have no knowledge of react and have been using vuejs, maybe we can email or talk over slack to work something out? cheers

pierotofy, 2018-01-21:
So https://github.com/expo/expo-three/pull/13 has been merged, which means work on this can go forward.

The reason for using React is not because of a choice of JS framework, it's because it's a React Native application, which means we can easily use three.js and Apple's ARKit (and someday soon possibly Android's ARCore).

To my knowledge there's no React Native equivalent for vuejs.

pierotofy, 2020-04-02:
I'm closing this since I have no bandwidth to implement it and priorities have changed.

closed 2020-04-02 by pierotofy