WebODM/WebODM #317: Processing error 1 message

DroneMapper, 2017-11-08:
8724	Increase of energy - stopping optimization
	Took: 295.437s
Generating texture patches:
	Running... 
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 124, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

Exception: Child returned 137

pierotofy, 2017-11-08:
Could you please follow this [template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md) when opening an issue? Thank you 👍

DroneMapper, 2017-11-10:
Installed it with docker toolbox
using firefox
processing ends with Process exited with code 1

Options: min-num-features: 4000, use-pmvs: true
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mage: DJI_0343.JPG 00000134
Image: DJI_029*********************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************Killed
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/cmvs.py", line 57, in process
    system.run('{bin} {prefix}/ {max_images} {cores}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

pierotofy, 2017-11-10:
A `Killed` message usually means you ran out of memory. What happens if you turn off the `use-pmvs` flag?

pierotofy, 2017-11-10:
Related to the first log: https://github.com/OpenDroneMap/OpenDroneMap/issues/706

DroneMapper, 2017-11-12:
I tried running the same process with Options: min-num-features: 10000, use-pmvs: true

The timer keeps running but the process stop with the error "set a processing node"

There are no usable processing nodes. We tried to reach:

    node-odm-1:3000 (queue: 1)

Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.

DroneMapper, 2017-11-12:
I restarted everything, the "set a processing node" error is gone but the timer still runs and the little wheel keeps on spinning, but nothing happens. There only a blinking of the last line as below, but that is where it stops

2017-11-12 16:22:07,760 DJI_0117.jpg resection inliers: 967 / 1124
2017-11-12 16:22:07,790 Adding DJI_0117.jpg to the reconstruction
2017-11-12 16:24:28,597 Ceres Solver Report: Iterations: 8, Initial cost: 5.362052e+03, Final cost: 5.332873e+03, Termination: CONVERGENCE
2017-11-12 16:24:29,609 Bundle setup/run/teardown 3.47509503365/137.122366905/1.01431107521
2017-11-12 16:24:29,912 Removed outliers: 1
2017-11-12 16:24:31,810 -------------------------------------------------------
2017-11-12 16:24:31,840 DJI_0118.JPG resection inliers: 752 / 844
2017-11-12 16:24:31,862 Adding DJI_0118.JPG to the reconstruction

pierotofy, 2018-03-02:
If there are further questions about this subject,  according to the [guidelines](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md) please let's move them to the forum. http://community.opendronemap.org/c/webodm

closed 2018-03-02 by pierotofy