WebODM/WebODM #322: DSM+DTM Option Error on Native WebODM

ybh0616, 2017-11-12:
### How did you install WebODM? (Docker, natively, ...)?
I'm running WebODM natively on Ubuntu 16.04 LTS.

### What's your browser and operating system?
whatismybrowser.com/w/KG4SGAS

### What is the problem?
The DSM+DTM option is not applied on WebODM.

### What should be the expected behavior?
### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
In my opinion, instructions on setting up lidar2dems need to be provided when installing WebODM natively. I can't find a guide on it at this moment. I'm always grateful for your great works.

pierotofy, 2017-11-12:
Hi @ybh0616, lidar2dems is not part of WebODM, it's installed as part of opendronemap. Perhaps you need to update your processing node?

ybh0616, 2017-11-12:
Oh, I see. Sorry. I should check it out. Thank you for your quick reply.

closed 2017-11-13 by pierotofy